mne-tools/mne-python #8779: show parameter doesn't work with evoked.plot_topomap

jasmainak, 2021-01-23:
Here is the script to reproduce:

```py
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=120)

events = mne.find_events(raw, stim_channel='STI 014')
event_dict = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
              'visual/right': 4, 'face': 5, 'button': 32}
epochs = mne.Epochs(raw, events, tmin=-0.2, tmax=0.5, event_id=event_dict,
                    preload=True)
epochs.pick_types(eeg=True, meg=False)
evoked = epochs.average()
evoked.plot_topomap(show=False)
```

jasmainak, 2021-01-23:
same issue seems to exist with `evoked.plot_white`

agramfort, 2021-01-24:
for me when running this code from a script It does not open a window until
I call plt.show()

now if I paste it is shown but it's just how ipython works.

>

jasmainak, 2021-01-24:
okay fair enough. Didn't realize that you need to do `$ ipython` without `--matplotlib` flag for this to work. Thanks for checking

closed 2021-01-24 by jasmainak