mne-tools/mne-python #8784: LinAlgError: SVD did not converge when plotting data (raw.plot())

FriederikeScholten, 2021-01-25:
I am currently going through the tutorials provided by MNE and when I try to use the built-in plot function, I often get an error that the SVD did not converge (see attached picture). Two other colleagues have the same problem (using Spyder & Visual Studio Code). One of them made the experience that it does work sometimes after re-running the cell multiple times. Any idea what the issue might be here? 
[image]

agramfort, 2021-01-25:
with what data is that?

FriederikeScholten, 2021-01-25:
@agramfort , we used the 'sample' dataset that is used in the tutorial

agramfort, 2021-01-25:
what does

mne.sys_info()

give you?

>

larsoner, 2021-01-25:
If you're using NumPy with OpenBLAS it could be https://github.com/xianyi/OpenBLAS/issues/3044 which won't be fixed until an OpenBLAS release then NumPy release. Or if you're using NumPy with MKL you could be caught by https://community.intel.com/t5/Intel-Distribution-for-Python/Bug-in-GESDD-but-not-GESVD/m-p/1105241 which has been around for 5 years and stopped getting replies from Intel folks. We can work around either one probably with a `from mne.fixes import _safe_svd` and using that instead of `scipy.linalg.svd`, can you give it a try? If it works, it would be awesome if you could open a quick PR for this!

If it does not fix the problem, you could be hitting this issue because you have `nan` or `inf` in your projector, and we should add some check for the projector being invalid

FriederikeScholten, 2021-01-25:
This is what the mne.sys_info() gives me: 
[image]

larsoner, 2021-01-25:
@FriederikeScholten can you double check that `all(np.isfinite(proj['data']['data']).all() for proj in raw.info['projs'])` evaluates to `True`? If it is then you're probably hitting the [Intel bug](https://community.intel.com/t5/Intel-Distribution-for-Python/Bug-in-GESDD-but-not-GESVD/m-p/1105241). If you can [check your CPU architecture](https://www.howtogeek.com/413942/how-to-see-what-cpu-is-in-your-pc-and-how-fast-it-is/#:~:text=Head%20to%20Control%20Panel%20%3E%20System,Processor%E2%80%9D%20under%20the%20System%20heading), try out some of the cases I posted in that issue, and tell them you're hitting the issue, too? Maybe more pings/reports will get them to fix it.

And in the meantime, feel free to try using `_safe_svd`, it should work around the issue

FriederikeScholten, 2021-03-03:
Thanks for the suggestions. I do not know how (to my knowledge I did not change anything), but the plot function works fine now, also for my colleague. Just to follow up, the code: all(np.isfinite(proj['data']['data']).all() for proj in raw.info['projs']) evaluates True.

closed 2021-03-03 by FriederikeScholten