mne-tools/mne-python #8786: MNE website down?

cbrnr, 2021-01-25:
I hope this isn't related to our recent renaming of master -> main, but our website at https://mne.tools is currently down. Any ideas?

hoechenberger, 2021-01-25:
Have you tried reloading? I witnessed it only showing content upon the second losing attempt. Thought this was an issue got me locally only

hoechenberger, 2021-01-25:
Oh it's a 404. Boni did not have this until a few hours ago (last time I checked)

cbrnr, 2021-01-25:
😱 at least we didn't break it, but still... 😥

hoechenberger, 2021-01-25:
> 😱 at least we didn't break it, but still... 😥 

Or maybe we did and there's some kind of hysteresis due to caching somewhere 😅🤯

larsoner, 2021-01-25:
I had to rename `master` to `main` for the `mne-tools.github.io` repo, and this broke things temporarily probably due to some GitHub bug. Pushing an empty commit after the last `deploy` step seems to have fixed it.

closed 2021-01-25 by larsoner

hoechenberger, 2021-01-25:
Thanks for the swift action, @larsoner!!