mne-tools/mne-python #8791: Inconsistent and probably problematic behaviors in coregistration

yh-luo, 2021-01-26:
#### Describe the bug
Hi, I've noticed some inconsistent and probably problematic behaviors in ``mne coreg``. The problems are recorded in [the video](https://drive.google.com/file/d/11Ydv1H63JJWnbz6Rc92DLKfi5SMflWNA/view?usp=sharing)
1. The values in **MRI Fiducials** are not always updated when the user clicks on different positions. For example, the coordinates of nasion are *not* updated in ``00:02-00:09`` but that of LPA *are* updated in ``00:12-00:17``. Then, the coordinates of RPA are *not* updated in ``00:17-00:20``. It's really frustrating to understand what's happening...
2. When the values in **MRI Fiducials** are provided by the user, the coordinate system seems to be inverted. The weird behavior can be viewed from ``00:25`` to the end of the video.

Because the MRI images were acquired in a fMRI experiment which required the participants to wear goggles, sometimes I need to use FreeView to find the possible coordinates for better coregistration results. The above problems make it hard to do that.

Nothing was shown on the terminal running ``mne coreg``.

#### Expected results
Reasonable coregistration.

#### Actual results
Confusing behaviors in ``mne coreg``

#### Additional information
OS: Ubuntu 20.04
mne.sys_info()
```
Platform:      Linux-5.8.0-40-generic-x86_64-with-glibc2.10
Python:        3.8.5 (default, Sep  4 2020, 07:30:14)  [GCC 7.3.0]
Executable:    /home/ntubabylab/anaconda3/envs/mne/bin/python
CPU:           x86_64: 12 cores
Memory:        62.8 GB

mne:           0.22.0
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
nilearn:       Not found
dipy:          Not found
cupy:          8.1.0
pandas:        1.1.3
mayavi:        4.7.2
pyvista:       0.26.1 {pyvistaqt=0.2.0, OpenGL 4.5.0 NVIDIA 450.102.04 via GeForce GTX 1660/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.1
```

larsoner, 2021-01-26:
To me everything looks okay/correct until around the 40 mark at which point you set the head opacity to zero.

When you set the opacity to zero what you end up seeing is the *inside* of the head, which is still rendered with opacity 1. This leads to the confusing interaction. This is actually intended behavior so that dig points behind that point are occluded, but should probably be disabled during the LPA/Nasion/RPA step.

I make this surface disappear when in fiducials mode and add an option to turn it on/off in standard mode in #8793

mmagnuski, 2021-01-26:
@larsoner I would also expect the nasion coordinates to be updated after each click not only after switching to another fiducial point and then back to nasion.

yh-luo, 2021-01-27:
If that is the inside of the head that makes sense now. But still it is a bit annoying that the coordinates of nasion are not always updated, especially when I try to manually set the value. I tested few more times, the coordinates of LPA and RPA do not show this behavior.

larsoner, 2021-01-27:
I pushed a commit to fix the updating in #8793 can you give it a try?

yh-luo, 2021-01-28:
@larsoner it solves the problems and works smoothly, thank you!

PR #8793 (2021-01-28): MRG, ENH: Add inner surface option to coreg
- Add option to control appearance of opaque inside surface of the head to mne coreg
- Make the opaque surface always disappear in fiducials mode (it's not ever really helpful there)

Closes #8791
closed 2021-01-28 by agramfort