mne-tools/mne-python #8795: EDF Reader memory hungry

agramfort, 2021-01-27:
try `mprof run` on this script:

```
import mne
mne.set_log_level("WARNING")

def read_raws(paths, preload):
    return [mne.io.read_raw_edf(path, preload=preload) for path in paths]

paths = mne.datasets.sleep_physionet.age.fetch_data(
    subjects=range(1, 10), recording=[1])
paths, _ = zip(*paths)

raws = read_raws(paths, preload=False)
```

it produces

[image]

it's very surprising that it requires so much memory even when preload=False.

see https://github.com/braindecode/braindecode/pull/188

cc @gemeinl

larsoner, 2021-01-27:
Locally with your script I see:

[image]

Here is a more memory-efficient version, created by allowing each to GC as it's created:
```
import mne
mne.set_log_level("WARNING")

def read_raws(paths, preload):
    raw = None
    for path in paths:
        this_raw = mne.io.read_raw_edf(path, preload=preload)
        if raw is None:
            this_raw = raw
        else:
            raw.append(this_raw)

paths = mne.datasets.sleep_physionet.age.fetch_data(
    subjects=range(1, 10), recording=[1])
paths, _ = zip(*paths)

raws = read_raws(paths, preload=False)
```

[image]

So one option is to concat so there is less overhead. But if ~80 MB per instance is too onerous *and* concatenation is not an option then I can look deeper....

larsoner, 2021-01-27:
For `raws[0]` I see:
```
>>> (len(raws[0].times), raws[0].times.nbytes // 1024 // 1024)
(8406000, 64)
```
So it takes up 64 MB to have `raw.times`.

In theory we could compute this on the fly when necessary but we would have to be careful about not introducing performance regressions.

cbrnr, 2021-01-27:
It is even much simpler to save memory: replace the list comprehension with a generator expression, this has exactly the same much reduces memory consumption than the modified version from @larsoner.

Regarding the relatively large `times` array - would it be reasonable to compute it on the fly only for `preload=False`?

cbrnr, 2021-01-27:
Oh wait, with a generator expression it's even better (at least on my machine):

[image]

cbrnr, 2021-01-27:
Maybe ignore my comment on using a generator expression, I assume the goal here is to have all raws in memory...

larsoner, 2021-01-27:
Looking back it can be a 400x slowdown to create on the fly apparently

https://github.com/mne-tools/mne-python/pull/2126

I wouldn't want to force this on people who use `preload=False`. Maybe we can improve it with better coding but first let's see if @agramfort thinks it's indeed necessary...

agramfort, 2021-01-27:
the use case from deeplearning workflow where we want to read from disk when needed windows of data from disk. There are thousands of EEG files... I'll let @gemeinl comment further as he raised this issue first.

cbrnr, 2021-01-27:
Yes, I think it really depends on the use case. However, why does the initial script consume >600MB when `preload=False`? Is it the `times` array (one for each of the 9 files)? Why is this not the same for the appended version (`times` will be appended instead of separate)?

agramfort, 2021-01-27:
Yes I think there is something to understand. It's indeed suspicious

>

gemeinl, 2021-01-28:
Thanks for looking into this!
Fitting more raws into RAM would be very convenient but is not required from our side.
In a typical setting, with a very large dataset that does not fit entirely into RAM, we will read raw signals, preprocess them, epoch them, and store them to disk.
For training, we will then read epochs with `mne.read_epochs` with `preload=False` and load individual epochs with `get_data(item=index)`. So for us, it is important that this works efficiently and fast. So far, we have not observed those memory issues with `read_epochs` and made observations that loading should be fast enough (see https://github.com/braindecode/braindecode/pull/75).

agramfort, 2021-01-28:
@gemeinl can you be explicit and quantitative about what is current slow for your needs?
The last comment suggests there is no problem in the end...

gemeinl, 2021-01-28:
It depends on the deep learning model, on the GPU, on the batch size as well as on the dataset.
@hubertjb has done comparisons of an in-memory and a lazy loaded dataset and found minor slowdowns. @hubertjb Do you remember a factor of slowdown or rough estimates of model training time and lazy loading time in your specific setting?

gemeinl, 2021-01-28:
If it plays a role in your consideration whether to compute the times on-the-fly or not, I could achieve ~50% speed up generating the times. See:
```
import time
import numpy as np
import pandas as pd

sfreq = 256.
results = pd.DataFrame()
for n_times in [1e3, 1e4, 1e5, 1e6]:
    for i in range(1000):
        start = time.time()
        np.arange(n_times) / sfreq
        t = time.time() - start
        results = results.append(pd.Series(
            {"n_times": n_times, "kind": "mne", "time": t, "sfreq": sfreq}
        ), ignore_index=True)
        start = time.time()
        np.arange(0, n_times/sfreq, 1/sfreq)
        t = time.time() - start
        results = results.append(pd.Series(
            {"n_times": n_times, "kind": "custom", "time": t, "sfreq": sfreq}
        ), ignore_index=True)

results.groupby(["sfreq", "n_times", "kind"]).median()
```
[image]

Executed on an AMD Ryzen 5 4600h.

closed 2021-02-05 by larsoner