mne-tools/mne-python #8797: discrepancy when plotting vertex time course with pyvista backend

rythorpe, 2021-01-27:
#### Describe the bug
The MNE time course looks different when plotting a vertex within the brain visualization GUI via the pyvista backend (top figure shown below) compared to manually plotting the vertex using ``SourceEstimate.data`` (bottom figure shown below).

#### Steps to reproduce
See https://gist.github.com/rythorpe/3c57fdb572bc17bb1f3e4bb70b3e1104 using the ``somato`` dataset. Selecting the exact vertex within the brain visualization window can be tricky (unless there is a more controlled way to to this through the API?).

#### Expected results
I expected the vertex time courses rendered in the GUI to match those plotted manually with matplotlib. Furthermore, I expected the vertex of maximal activity (blue trace shown below) to reflect the largest peak (compared to the orange trace) in the GUI.

#### Actual results
Note that the blue trace represents the time course of the right hemisphere vertex retrieved with `SourceEstimate.get_peak()` and that the vertex IDs are the same in each figure.

[image]

[image]

jasmainak, 2021-01-27:
Thanks a lot @rythorpe for the bug report. It does looks like something funky is going on. @GuillaumeFavelier do you know where we should look in the code for this? @rythorpe mentioned to me that he would be up for doing a PR

larsoner, 2021-01-27:
Do you still have this problem with smoothing_steps='nearest'? The time course plotted is for the smoothed data.

rythorpe, 2021-01-28:
> Do you still have this problem with smoothing_steps='nearest'? The time course plotted is for the smoothed data.

That appears to do the trick (see plot below)! What is the motivation behind plotting the time course for smoothed data by default? Unless there is a strong use case for reducing high-frequency noise in the inverse solution, I feel like this can give a misleading first-impression of relative vertex time courses.

[image]

jasmainak, 2021-01-28:
> Furthermore, I expected the vertex of maximal activity (blue trace shown below) to reflect the largest peak (compared to the orange trace) in the GUI.

Wasn't this also an issue? Is this also fixed by changing the smoothing steps?

jasmainak, 2021-01-28:
> What is the motivation behind plotting the time course for smoothed data by default?

The source space is a decimated surface, so to get a proper visualization you need to smooth the result. You could also plot directly on the source space but since it's so decimated it will look very weird

rythorpe, 2021-01-28:
> > Furthermore, I expected the vertex of maximal activity (blue trace shown below) to reflect the largest peak (compared to the orange trace) in the GUI.
> 
> Wasn't this also an issue? Is this also fixed by changing the smoothing steps?

This is fixed by using `smoothing_steps='nearest'`. Note in the figure attached to my previous comment that the peak at t=0.041 sec in the blue trace is now greater in magnitude than in the orange trace.

jasmainak, 2021-01-28:
okay feel free to close issue then.

rythorpe, 2021-01-28:
Thanks for the help @larsoner and @jasmainak!

closed 2021-01-28 by rythorpe