mne-tools/mne-python #8798: defining backend in SourceEstimate.plot() sets default backend for subsequent renders

rythorpe, 2021-01-27:
#### Describe the bug
Explicitly defining a 3d rendering backend in `SourceEstimate.plot()` appears to set the backend for subsequent calls to the same method that don't explicitly define the backend.

#### Steps to reproduce
Run [this script](https://gist.github.com/rythorpe/0164d059eaab74c155c5246f2679f895) in an ipython shell.

#### Expected results
Explicitly defining the `backend` arg in `SourceEstimate.plot()` should open a window using the specified backend, and subsequent calls to `SourceEstimate.plot()` that don't define the `backend` arg should use the default (i.e., pyvista) backend as noted in the [documentation](https://mne.tools/stable/generated/mne.SourceEstimate.html?highlight=plot#mne.SourceEstimate.plot).

#### Actual results
If `backend='mayavi'` is explicitly defined in the beginning, subsequent calls to plot the source estimate will use the mayavi backend by default.

#### Additional information
Platform:      Linux-5.4.0-64-generic-x86_64-with-glibc2.10
Python:        3.8.6 | packaged by conda-forge | (default, Jan 25 2021, 23:21:18)  [GCC 9.3.0]
Executable:    /home/ryan/anaconda3/envs/hnn_core/bin/python
CPU:           x86_64: 12 cores
Memory:        15.5 GB

mne:           0.22.0
numpy:         1.19.5 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.6.0
matplotlib:    3.3.3 {backend=Qt5Agg}

sklearn:       0.24.1
numba:         0.52.0
nibabel:       3.2.1
nilearn:       0.7.0
dipy:          1.3.0
cupy:          Not found
pandas:        1.2.1
mayavi:        4.7.2
pyvista:       0.27.4 {pyvistaqt=0.3.0, OpenGL 4.5.0 NVIDIA 450.102.04 via Quadro P1000/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.12.3

agramfort, 2021-01-30:
cc @GuillaumeFavelier

GuillaumeFavelier, 2021-02-01:
The `backend` parameter in `plot_stc` does not play nicely with the 3d backend. I'll see how it can be fixed.

closed 2021-02-02 by GuillaumeFavelier