mne-tools/mne-python #8812: mne.viz.set_3d_view with pyvista backhand is not consistent!

dasdiptyajit, 2021-01-29:
I'm trying to do something similar to :
https://mne.tools/stable/auto_examples/visualization/plot_mne_helmet.html#sphx-glr-auto-examples-visualization-plot-mne-helmet-py

#### Describe the bug
However, I am seeing an anti-aliasing effect during interactive mode in ipython and Jupyter-notebook

Expected results:
[image]

Getting: 
[image]

#### Additional information
Platform:      Linux-4.15.0-132-generic-x86_64-with-debian-buster-sid
Python:        3.7.9 (default, Aug 31 2020, 12:42:55)  [GCC 7.3.0]
Executable:    /home/diptyajit/anaconda3/bin/python
CPU:           x86_64: 4 cores
Memory:        15.6 GB

mne:           0.22.0
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.0.0
nilearn:       0.6.1
dipy:          1.1.0
cupy:          Not found
pandas:        1.2.0
mayavi:        4.7.2
pyvista:       0.27.4 {pyvistaqt=0.2.0, OpenGL 3.3 (Core Profile) Mesa 19.2.8 via Mesa DRI Intel(R) HD Graphics 530 (Skylake GT2)}
vtk:           9.0.1
PyQt5:         5.14.1

larsoner, 2021-01-29:
Can you write a little code snippet using `sample` data that replicates this? Or is it enough just to run the example above in Python and then in Jupyter to see different outputs?

dasdiptyajit, 2021-01-29:
# replicate the issue with Jupyter
# code

import os.path as op
import mne

sample_path = mne.datasets.sample.data_path()
subjects_dir = op.join(sample_path, 'subjects')
fname_evoked = op.join(sample_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
fname_inv = op.join(sample_path, 'MEG', 'sample',
                    'sample_audvis-meg-oct-6-meg-inv.fif')
fname_trans = op.join(sample_path, 'MEG', 'sample',
                      'sample_audvis_raw-trans.fif')
inv = mne.minimum_norm.read_inverse_operator(fname_inv)
evoked = mne.read_evokeds(fname_evoked, baseline=(None, 0),
                          proj=True, verbose=False, condition='Left Auditory')
maps = mne.make_field_map(evoked, trans=fname_trans, ch_type='meg',
                          subject='sample', subjects_dir=subjects_dir)
time = 0.083
fig = mne.viz.create_3d_figure((256, 256))
mne.viz.plot_alignment(
    evoked.info, subject='sample', subjects_dir=subjects_dir, fig=fig,
    trans=fname_trans, meg=False, eeg=False, surfaces='white',
    coord_frame='mri')
evoked.plot_field(maps, time=time, fig=fig, time_label=None)
mne.viz.set_3d_view(fig, azimuth=0, elevation=0, focalpoint=(0., 0., 0.), distance=1)

Results: 
[image]
[image]

GuillaumeFavelier, 2021-01-29:
This bug is described in https://github.com/mne-tools/mne-python/issues/7599 and is related to VTK9. [It has been patched for `Brain`](https://github.com/mne-tools/mne-python/pull/8576) but can probably occur on other viz that do not use the same algorithm. I think it's your case. I happen to work on porting evoked field map feature to use `Brain` instead. You can track the progress of this work in https://github.com/mne-tools/mne-python/pull/8749

closed 2022-05-24 by larsoner