mne-tools/mne-python #8814: Epochs.reject_tmin and reject_tmax aren't being adjusted for deviations due to sampling rate

hoechenberger, 2021-01-30:
MWE:

```python
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')

raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=60)
events = mne.find_events(raw, stim_channel='STI 014')

tmin, tmax = -0.2, 0.8
epochs = mne.Epochs(raw, events,
                    tmin=tmin, tmax=tmax,
                    reject_tmin=tmin, reject_tmax=tmax)

print(epochs.tmin)
print(epochs.reject_tmin)
print(epochs.tmax)
print(epochs.reject_tmax)
```

Output:
```python
-0.19979521315838786
-0.2
0.7991808526335514
0.8
```

Saving above Epochs to disk and trying to load them will fail:
```python
epochs_fname = '/tmp/epochs-epo.fif'

epochs.save(epochs_fname)
epochs_loaded = mne.read_epochs(epochs_fname)
```

Output:
```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Untitled-1 in 
     18 epochs.save(epochs_fname, overwrite=True)
     19 
---> 20 epochs_read = mne.read_epochs(epochs_fname)
     21 

<decorator-gen-212> in read_epochs(fname, proj, preload, verbose)

~/Development/mne-python/mne/epochs.py in read_epochs(fname, proj, preload, verbose)
   2704         The epochs.
   2705     """
-> 2706     return EpochsFIF(fname, proj, preload, verbose)
   2707 
   2708 

<decorator-gen-213> in __init__(self, fname, proj, preload, verbose)

~/Development/mne-python/mne/epochs.py in __init__(self, fname, proj, preload, verbose)
   2819         # again, ensure we're retaining the baseline period originally loaded
   2820         # from disk without trying to re-apply baseline correction
-> 2821         super(EpochsFIF, self).__init__(
   2822             info, data, events, event_id, tmin, tmax, baseline=None, raw=raw,
   2823             proj=proj, preload_at_end=False, on_missing='ignore',

<decorator-gen-198> in __init__(self, info, data, events, event_id, tmin, tmax, baseline, raw, picks, reject, flat, decim, reject_tmin, reject_tmax, detrend, proj, on_missing, preload_at_end, selection, drop_log, filename, metadata, event_repeated, verbose)

~/Development/mne-python/mne/epochs.py in __init__(***failed resolving arguments***)
    481         # check reject_tmin and reject_tmax
    482         if (reject_tmin is not None) and (reject_tmin < tmin):
--> 483             raise ValueError("reject_tmin needs to be None or >= tmin")
    484         if (reject_tmax is not None) and (reject_tmax > tmax):
    485             raise ValueError("reject_tmax needs to be None or <= tmax")

ValueError: reject_tmin needs to be None or >= tmin
```

agramfort, 2021-01-31:
can you have a look?

>

hoechenberger, 2021-01-31:
> can you have a look?

Yes I can work on this tomorrow.

PR #8821 (2021-02-02): MRG, BUG: Fix Epochs.reject_tmin, reject_tmax handling upon Epochs creation & cropping

- Adjust Epochs.reject_tmin, reject_tmax for time inaccuracies due to sfreq. Fixes #8814
- Adjust Epochs.reject_tmin, reject_tmax upon cropping. Fixes #8816
closed 2021-02-02 by hoechenberger