mne-tools/mne-python #8815: color argument is not respected in brain.add_label

jasmainak, 2021-01-31:
I tried the following:

```py
import os

import mne
from mne import read_source_estimate
from mne.datasets import sample

# Paths to example data
sample_dir_raw = sample.data_path()
sample_dir = os.path.join(sample_dir_raw, 'MEG', 'sample')
subjects_dir = os.path.join(sample_dir_raw, 'subjects')

fname_stc = os.path.join(sample_dir, 'sample_audvis-meg')

stc = read_source_estimate(fname_stc, subject='sample')

# Define plotting parameters
surfer_kwargs = dict(
    hemi='lh', subjects_dir=subjects_dir,
    clim=dict(kind='value', lims=[8, 12, 15]), views='lateral',
    initial_time=0.09, time_unit='s', size=(800, 800),
    smoothing_steps=5)

# Plot surface
brain = stc.plot(**surfer_kwargs)

vertex_id = 72752
label = mne.grow_labels('sample', vertex_id, 20, 0, subjects_dir=subjects_dir)[0]
brain.add_label(label, color='green')
```

it gives me an orange label instead of green.

Furthermore, if I do:

```py
brain.remove_labels()
```

the label is not removed from the image. Not sharing screenshots because I have some rendering issue with these plots on my mac but you can verify/reproduce easily with the scripts above.

hoechenberger, 2021-01-31:
Tagging @GuillaumeFavelier because he's the expert here 😇

GuillaumeFavelier, 2021-02-01:
I can't reproduce the bug with orange label. For me, `color='green'`: 

[image]

<details>

```py
import mne
Brain = mne.viz.get_brain_class()

subjects_dir = mne.datasets.sample.data_path() + '/subjects'
labels = mne.read_labels_from_annot(
    'fsaverage', 'HCPMMP1', 'lh', subjects_dir=subjects_dir)

brain = Brain('fsaverage', 'lh', 'inflated', subjects_dir=subjects_dir,
              cortex='low_contrast', background='white', size=(800, 600))
aud_label = [label for label in labels if label.name == 'L_A1_ROI-lh'][0]
brain.add_label(aud_label, color="green")
```

</details>

But I did notice the issue with `remove_labels()`, good catch @jasmainak !

jasmainak, 2021-02-01:
yes I green label also with your code. But did you try the code above that I shared? :)

[image]

GuillaumeFavelier, 2021-02-01:
I ran it and I got the same now. This is embarassing :sweat_smile: 

I'm on it

closed 2021-02-01 by larsoner