mne-tools/mne-python #8825: Incorrect high-resolution head rendered in mne coreg gui

yh-luo, 2021-02-02:
#### Describe the bug
Someone else's high-resolution head was used for the selected subject.

`lh.seghead` in FreeView looks like:
[image]
 
But the high-resolution head rendered in `mne coreg` gui looks like:
(Notice the differences of the top-left corner of the goggle, the ears, and the nose bridge.)
[image]

The high-resolution head also did not align to the head. No rotations were made, the head looks like:
[image]

I checked in FreeView that `lh.seghead` and `outer_skin.surf` were aligned.
From the above images, it seemed to me that someone else's dense head was rendered. The error messages indicated that `subject-102`'s dense head was used instead?
```python
mne coreg -s subject-346
Exception occurred in traits notification handler for object: <mne.gui._file_traits.SurfaceSource object at 0x7f3db86bcb30>, trait: file, old value: subjects//subject-102/bem/subject-102-head-dense.fif, new value: subjects//subject-346/bem/subject-346-head-dense.fif
Traceback (most recent call last):
  File "/home/ntubabylab/anaconda3/envs/mne/lib/python3.8/site-packages/traits/trait_notifiers.py", line 522, in _dispatch_change_event
    self.dispatch(handler, *args)
  File "/home/ntubabylab/anaconda3/envs/mne/lib/python3.8/site-packages/traits/trait_notifiers.py", line 484, in dispatch
    handler(*args)
  File "/home/ntubabylab/anaconda3/envs/mne/lib/python3.8/site-packages/mne/gui/_file_traits.py", line 175, in read_file
    bem = read_bem_surfaces(self.file, verbose=False)[0]
  File "<decorator-gen-98>", line 22, in read_bem_surfaces
  File "/home/ntubabylab/anaconda3/envs/mne/lib/python3.8/site-packages/mne/bem.py", line 1270, in read_bem_surfaces
    _check_complete_surface(this)
  File "/home/ntubabylab/anaconda3/envs/mne/lib/python3.8/site-packages/mne/bem.py", line 270, in _check_complete_surface
    raise RuntimeError(msg)
RuntimeError: Surface outer skin  has topological defects: 13 / 383510 vertices have fewer than three neighboring triangles [44385, 44387, 44389, 45198, 46031, 46033, 46861, 46862, 47708, 48546, 49428, 50338, 53081]
```

#### Steps to reproduce
I can't provide a MWE because it only happened to this subject.

#### Expected results
subject-346's `lh.seghead`

#### Actual results
subject-102(?)'s `lh.seghead`

#### Additional information

The above images in this post will be removed in the future.

```
Platform:      Linux-5.8.0-40-generic-x86_64-with-glibc2.10
Python:        3.8.5 (default, Sep  4 2020, 07:30:14)  [GCC 7.3.0]
Executable:    /home/ntubabylab/anaconda3/envs/mne/bin/python
CPU:           x86_64: 12 cores
Memory:        62.8 GB

mne:           0.22.0
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
nilearn:       Not found
dipy:          Not found
cupy:          8.1.0
pandas:        1.1.3
mayavi:        4.7.2
pyvista:       0.26.1 {pyvistaqt=0.2.0, OpenGL 4.5.0 NVIDIA 450.102.04 via GeForce GTX 1660/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.1
```
[... 262 lines cut ...]
Surfaces passed the basic topology checks.
4. Creating sparse tessellation...
4.1 Decimating the dense tessellation...
4.2 Creating /mnt/sdb1/MEG_Prosody/subjects/subject-346/bem/subject-346-head-sparse.fif
outer skin  CM is  17.12  24.57 -62.60 mm
Surfaces passed the basic topology checks.
/home/ntubabylab/anaconda3/envs/mne/bin/mne:8: RuntimeWarning: Surface outer skin  has topological defects: 13 / 383510 vertices have fewer than three neighboring triangles [44385, 44387, 44389, 45198, 46031, 46033, 46861, 46862, 47708, 48546, 49428, 50338, 53081]

Consider using --force as an additional input parameter.
  sys.exit(main())
/home/ntubabylab/anaconda3/envs/mne/bin/mne:8: RuntimeWarning: Surface outer skin  is not complete (sum of solid angles yielded 0.999793, should be 1.)
  sys.exit(main())
/home/ntubabylab/anaconda3/envs/mne/bin/mne:8: RuntimeWarning: Surface outer skin  has topological defects: 53 / 30000 vertices have fewer than three neighboring triangles [1659, 1663, 1690, 1732, 1734, 1767, 1812, 1813, 1815, 1849, 1850, 1887, 2012, 2127, 2212, 2272, 2507, 3476, 5047, 5435, 6169, 6296, 6297, 6532, 6540, 6587, 6976, 7295, 7855, 8106, 8199, 8528, 8533, 8538, 8727, 8795, 8953, 9039, 10498, 10546, 11537, 11560, 11755, 12293, 12587, 13594, 13779, 14204, 14207, 14300, 14313, 14463, 14532]

Consider using --force as an additional input parameter.
  sys.exit(main())
/home/ntubabylab/anaconda3/envs/mne/bin/mne:8: RuntimeWarning: Surface outer skin  is not complete (sum of solid angles yielded 0.999567, should be 1.)
  sys.exit(main())
/home/ntubabylab/anaconda3/envs/mne/bin/mne:8: RuntimeWarning: Surface outer skin  has topological defects: 17 / 2500 vertices have fewer than three neighboring triangles [176, 180, 184, 189, 271, 272, 677, 775, 991, 1060, 1061, 1107, 1128, 1160, 1161, 1162, 1190]

Consider using --force as an additional input parameter.
  sys.exit(main())
/home/ntubabylab/anaconda3/envs/mne/bin/mne:8: RuntimeWarning: Surface outer skin  is not complete (sum of solid angles yielded 1.00046, should be 1.)
  sys.exit(main())
```

agramfort, 2021-02-02:
Did you track down what files are used to see where it can go wrong?
it's going to be hard to debug without access to the data...

>

yh-luo, 2021-02-02:
@agramfort I think all the files are fine. They look correct in FreeView. That's why I suspect the problem is from the rendering process.

If I remember correctly, `bem/subject-head.fif` (head in coregistration gui) is converted from `bem/subject_outer_skin_surface`, and `bem/subject-head-dense.fif` (high-resolution head) is converted from `subject/mri/lh.seghead`
Both `subject/mri/lh.seghead` and `subject_outer_skin_surface` looks correct in FreeView (the first image, `subject_outer_skin_surface` not showing).

From the first line in the error messages, I think the problem is gui related. `subject-346` was the specified subject but `subject-102`'s file (another subject) was mentioned, which is weird.
```python
Exception occurred in traits notification handler for object: <mne.gui._file_traits.SurfaceSource object at 0x7f3db86bcb30>, trait: file, old value: subjects//subject-102/bem/subject-102-head-dense.fif, new value: subjects//subject-346/bem/subject-346-head-dense.fif
```

agramfort, 2021-02-02:
I would look into the fif files you have to see if the information on
subject they contain is correct

>

yh-luo, 2021-02-02:
I'm not that familiar with the fif files sadly... [`subjects/subject-346/bem`](https://drive.google.com/file/d/100qqKx73ymQ-9G6I0yKocqH9RtQaGZun/view?usp=sharing) can be downloaded via the link.

agramfort, 2021-02-02:
you can use:

In [25]: import mne

In [26]: pos, tri = mne.read_surface('outer_skin.surf')

In [27]: from mayavi import mlab

In [28]: mlab.triangular_mesh(*pos.T, tri)
Out[28]: <mayavi.modules.surface.Surface at 0x7fe64be30720>

to plot the .surf files. For the fif bem files you can use:

In [37]: surf = mne.bem.read_bem_surfaces('subject-346-head.fif')[0]

In [38]: mlab.triangular_mesh(*surf['rr'].T, surf['tris'])

I let you dig in

larsoner, 2021-02-02:
> I think all the files are fine. They look correct in FreeView

Based on this output:
```
RuntimeError: Surface outer skin  has topological defects: 13 / 383510 vertices have fewer than three neighboring triangles [44385, 44387, 44389, 45198, 46031, 46033, 46861, 46862, 47708, 48546, 49428, 50338, 53081]
```
I don't think the surfaces are currently usable. They might look fine in FreeView, but `mne coreg` (and other parts of MNE code) rely on the surface being *topologically correct*, and it's not. A topologically incorrect surface (with holes, missing neighbors, etc.) can look just fine by eye.

The coreg error is that when it tries to switch surfaces between subjects it fails, leaving things in an inconsistent state. That's a bug, but if we fix your surfaces it'll get you going again and avoid it. And even if we make the failure more graceful then you'll still need to fix the surface for that subject in order to be able to process them, so let's work on that first.

As a first step, you can try using the code here:

https://github.com/christian-oreilly/infant_template_paper/blob/master/template_building.py#L186-L204

It has some dependencies like pymeshfix and trimesh that you'll have to install, but basically I would 1) backup the original files, 2) read them with `read_surface` or `read_bem_surfaces`, 3) fix them using this function, and then 4) write them back out with `write_surface` or some code [hacked together from our private bem functions](https://github.com/mne-tools/mne-python/blob/4df9ed838f1d56471d071308fb62b9ec749cb10d/mne/bem.py#L1200-L1202) (note: using private functions is not a stable long-term solution as the API can change at any time).

In the meantime I could work on a PR to make these mesh fixing functions public (albeit requiring these packages) and then add a `write_head_bem` or so to allow people to write out custom `head.fif` files.

yh-luo, 2021-02-02:
@larsoner thanks for the explanation. Yes, there are holes in the head eyeing from FreeView.

I understand the importance of BEM surfaces and follow the amazing [Blender tutorial](https://mne.tools/stable/auto_tutorials/source-modeling/plot_fix_bem_in_blender.html) to fix a lot of surfaces, but only for `brain`, `inner_skull` and `outer_skull` which I think will be used in MEG source reconstruction (`inner_skull` to be specific).

I wonder if `head.fif` and `head-dense.fif` need to be fixed too. That would be a painful job because nearly all my subjects have topological defects in the `head-dense.fif` due to wearing goggles and headphones in the scan. The data were collected before I'm in the loop so 🤦 
Another thing intrigued me is that `mne coreg` only complains this subject, but not all the subjects.

If I understand correctly, these head files are only used for coregistration? My current workaround is to use FreeView to view the T1 images and decide the possible MNI coordinates for fiducial points and use that in the gui. Then, the usual alignment process is followed. Do you think that's good enough for coregistration?

The code and the paper look great (infant template brains are life-saving!), I will try that!

larsoner, 2021-02-02:
> I wonder if head.fif and head-dense.fif need to be fixed too. 

Some of the computations even in `mne coreg` assume surfaces are topologically correct, such as the inside/outside coloring of points. Really it's best to fix the issues, and if we can make a function / use the functions from the infant MRI templates, then you can loop over all of your subjects and fix them all in one go without needing any manual intervention.

> If I understand correctly, these head files are only used for coregistration? My current workaround is to use FreeView to view the T1 images and decide the possible MNI coordinates for fiducial [...]

I would ensure that the actual correct subject surface is being used, otherwise it seems less reliable.

yh-luo, 2021-02-03:
It is easier than I previously thought thanks to the well-organized code.
I was able to fix(?) the high-resolution head using `fix_all_defects`. `mne coreg` accepts the `head-dense.fif` though the holes are still obvious. Not quite sure why...

Since it's more like a real-world problem than a bug, I'm closing the issue.

**lh.seghead in FreeView**
[image]
**After (`head-dense.fif` in `mne coreg`)**
[image]

Codes
```python
import os.path as op
import shutil

import mne
from mne.bem import _read_bem_surfaces_fif, _surfaces_to_bem, read_surface
from mne.io.constants import FIFF

from config import subjects_dir
from template_building import fix_all_defects

subject = 'subject-346'
bem_dir = op.join(subjects_dir, subject, 'bem')

# Load the surface
surf = _read_bem_surfaces_fif(op.join(bem_dir, f'{subject}-head-dense.fif'),
                              None)[0]

coords = surf['rr']
faces = surf['tris']

fix_coords, fixed_faces = fix_all_defects(coords, faces)

# get info
# _, _, volume_info = read_surface(op.join(bem_dir, 'watershed',
#                                          f'{subject}_outer_skin_surface'),
#                                  read_metadata=True)

# surf to bem for .fif
surf = _surfaces_to_bem([dict(rr=fix_coords, tris=fixed_faces)],
                        [FIFF.FIFFV_BEM_SURF_ID_HEAD], [1],
                        rescale=False)
fname_dense_head = op.join(bem_dir, f'{subject}-head-dense.fif')
# copy the original file
if op.isfile(fname_dense_head):
    shutil.copy(fname_dense_head,
                fname_dense_head.replace('dense', 'dense_orig'))

# Overwrite the surface
mne.write_bem_surfaces(fname_dense_head, surf, overwrite=True)
```

Essential `template_building.py` (functions that are not used or relying on unpublished codes are removed)
```python
import numpy as np
import pymeshfix
import trimesh
from mne.bem import _get_solids
from mne.surface import _triangle_neighbors

def fix_all_defects(vertices, faces_):
    if has_degenerated_faces(vertices, faces_):
        vertices, faces_ = remove_degenerated_faces(vertices, faces_)
        assert (not has_degenerated_faces(vertices, faces_))

    if has_topological_defects(vertices, faces_):
        print("The decimated mesh has topological defects. Fixing it.")
        vertices, faces_ = fix_topological_defects(vertices, faces_)
        if has_degenerated_faces(vertices, faces_):
[... 119 lines cut ...]
        new_face = np.delete(new_face, np.where(new_face == ind))
        assert (
            len(new_face) == 3
        )  # If == 4, it means that face1 and face2 do not share a common edge

        new_det = np.linalg.det(vertices[new_face])
        assert new_det  # If zero, the three points are colinear

        # Align the normals
        det1 = np.linalg.det(vertices[face1])
        if np.sign(det1) == np.sign(new_det):
            new_face = new_face[[1, 0, 2]]

        new_faces.append(new_face)

    return np.array(ind_faces_to_remove, dtype=int), new_faces

def reindex_vertices(vertices, faces_, ind_vertices_to_remove):
    decrement = np.cumsum(
        np.zeros(vertices.shape[0], dtype=int) +
        np.in1d(np.arange(vertices.shape[0]), ind_vertices_to_remove))
    vertices = np.delete(vertices, ind_vertices_to_remove, axis=0)
    faces_ = faces_ - decrement[faces_]
    return vertices, faces_
```

larsoner, 2021-02-03:
@yh-luo I'll reopen until we get a variant of this code in MNE-Python. I've also needed to use it so that's at least three people, and these topology questions come up a lot so I think it's worth pursuing proper inclusion as a function.

@christian-oreilly since you did all the heavy lifting on this, would you be up for opening a PR to add a `mne.surface.fix_topology(rr, tris)` function? Then I'm happy to hack away at the code to add tests, etc. to get it working, but the first commit will already get you code credit for the changes when we squash+merge, which I think is helpful since you did the heavy lifting!

I think in the end we probably also want a `write_head_bem` function since the code to do this currently requires poking around in private attributes. Then the fix could have maybe just been for you @yh-luo :
```
surf = mne.read_bem_surfaces(fname_head)[0]
surf['rr'], surf['tris'] = mne.surface.fix_topology(surf['rr'], surf['tris'])
mne.write_head_bem(fname_head, surf)
```

yh-luo, 2021-02-04:
@larsoner Another small thing is that `mne.read_bem_surfaces` complains about the topological defects and raise `RuntimeError`, that's why I used the private `_read_bem_surfaces_fif` to get around this.
To make your proposal work, an argument like `on_missing` in `mne.Epochs` (maybe `on_defects`) to let the user proceed to `fix_topology` would be nice? If so:
```python
surf = mne.read_bem_surfaces(fname_head, on_defects='warn')[0]
surf['rr'], surf['tris'] = mne.surface.fix_topology(surf['rr'], surf['tris'])
mne.write_head_bem(fname_head, surf)
```
```python
RuntimeWarning: Surface outer skin  has topological defects: 13 / 383510 vertices have fewer than three neighboring triangles [44385, 44387, 44389, 45198, 46031, 46033, 46861, 46862, 47708, 48546, 49428, 50338, 53081]
```
Both `write_head_bem` and this are easy to implement, I can make a PR for it if no one is already up to.

larsoner, 2021-02-04:
That would be great @yh-luo, please do!

christian-oreilly, 2021-02-04:
Yes, that would be @yh-luo ; I could have a look at this issue at some point but not in a near future because I need to do a few other things in priority at the moment. So if you can tackle it, that is perfect!

tillhabersetzer, 2022-03-01:
Thank you very much for providing the code. I had the same problem and it worked for me too! I would really appreciate to see this function `mne.surface.fix_topology(rr, tris)`  implemented in mne python