mne-tools/mne-python #8826: MNE-Python import time

cbrnr, 2021-02-02:
It seems to me like it's taking longer to `import mne` from version to version. Currently, importing takes a little over one second. I don't know if that's OK (probably) or if we should invest a little time to see if we can easily cut down on import time. I've already profiled the import with:

```
python -X importtime -c "import mne" 2> mne_import.log
```

`tuna` (`pip install tuna`) can be used to view the output interactively:

```
tuna mne_import.log
```

Here's a screenshot of the output for a rough overview (you have to run `tuna` yourself for an interactive version):
[image]

It seems like a lot of time is spent in `mne.io`, which maybe isn't necessary IDK.

Do you think this is worth pursuing or is a second of import time OK?

hoechenberger, 2021-02-02:
> `tuna` (`pip install tuna`) can be used to view the output interactively:

Neat!! Didn't know about that one, thanks for the hint! 🦄

agramfort, 2021-02-02:
how would you approach this issue? what makes you think we can do better?

cbrnr, 2021-02-02:
I would look into what happens in the various modules. I haven't checked at all because I wanted to discuss if this is even worth the effort. Subjectively, I feel that importing MNE takes very long compared to other packages (e.g. pandas takes about half as long), but I don't have any objective measurements.

agramfort, 2021-02-02:
if you install mne just with numpy, scipy available (which the only
required deps) how long does it take?
I think importing our deps can be a bottleneck

>

hoechenberger, 2021-02-02:
In PsychoPy this issue was addressed by using lots of lazy imports, e.g. `import psychopy.visual` wouldn't import all `visual` modules right away, but only once you actually tried to access them. A couple years ago, some IDEs had trouble extracting completion suggestions, but this may have been resolved now, I'm not sure.

hoechenberger, 2021-02-02:
> In PsychoPy this issue was addressed by using lots of lazy import

see e.g. https://github.com/psychopy/psychopy/blob/be80b2cbd431e17792a61ded91136d54ccc5d40a/psychopy/visual/__init__.py#L58

hoechenberger, 2021-02-02:
(I should add that I was never a fan of those lazy imports personally … this was just too much voodoo for my taste, and I don't mind imports taking a second or two, but that may just be me)

cbrnr, 2021-02-02:
There's no difference whether or not I have optional deps installed. I also think these lazy imports are a bit ugly and 1s probably doesn't warrant going in that direction.

The only other solution that comes to mind is not to import some submodules, but this will be a pretty huge breaking change and probably not worth it. Although we could argue that some more exotic/independent modules (e.g. report, datasets) do not need to be imported by `mne` directly.

larsoner, 2021-02-02:
> I think importing our deps can be a bottleneck

All optional deps already have their imports nested so they shouldn't be the culprit. There is a [test specifically for this](https://github.com/mne-tools/mne-python/blob/main/mne/tests/test_import_nesting.py#L16-L32) to prevent us from accidentally having them import when `import mne` is done.

One thing that might be less painful now that NumPy uses reasonable `linalg` functions is that we probably could nest all SciPy imports (and probably convert some `scipy.linalg.X` calls to `np.linalg.X` calls) without too much work.

`urllib.requests` takes some time and is only needed in one or two functions so we can nest it, too.

I can give some of these a shot and see if they bring some tangible benefits with `importtime-waterfall` (which is cool!)

larsoner, 2021-02-02:
Looks like on my system that it takes about 900 ms, and making `verbose` just `return function` cuts it down to about 600 ms. Most of this is due to our doc-filler dealing with indents, which I think I can make more efficient. I'll work on combining that with nesting some imports to see if I can get it below 500 ms at least.

cbrnr, 2021-02-02:
What do you mean by making `verbose` just `return function`? In any case, cutting it down by a third is already pretty awesome and definitely worth it!

PR #8829 (2021-02-07): MRG, ENH: Speed up import by caching indented params
So far 940 ms to 560 ms on my system by caching the indented versions of our docdict. It says 325 sec of this is importing numpy for me so not a ton left to be gained. I think there might be some other gains to be made for example by nesting scipy imports, though, so I'm at least going to try that.

Closes #8826
closed 2021-02-07 by agramfort