mne-tools/mne-python #8833: ENH: New interface for Coregistration

GuillaumeFavelier, 2021-02-03:
This PR summarizes the ideas to implement for `coreg`:

- [ ] Decouple GUI code from model (suggested in https://github.com/mne-tools/mne-python/pull/6693#issuecomment-542364434)
- [ ] Build a satisfying [app mockup](https://app.moqups.com)

hbk008, 2021-11-13:
I am not able to get past this: 

from mne.coreg import Coregistration

it throws error: cannot import name 'Coregistration' from 'mne.coreg' 

The other function from mne.coreg work well like getting fiducials from fiducials = mne.coreg.get_mni_fiducials(subject, subjects_dir=subjects_dir)
but this import Coregistration from mne.coreg fails for some reason. 
Can someone help me resolve this?

hoechenberger, 2021-11-13:
> from mne.coreg import Coregistration

To start up the coregistration GUI, you need to call

```python
mne.gui.coregistation()
```

hbk008, 2021-11-13:
Yes, but I want to use automated coregistration as explained in MNE tutorial: https://mne.tools/dev/auto_tutorials/forward/25_automated_coreg.html

So, how should I resolve this error?

hoechenberger, 2021-11-13:
@hbk008 I cannot reproduce this with MNE-Python 0.24. Please update to MNE-Python 0.24, then it should work.

Please note that the issue tracker is intended for bug reports and development issue tracking only. For usage questions like yours, please refer to our forum at https://mne.discourse.group

hbk008, 2021-11-13:
Okay, will post in the forum next time. You meant update from 0.23 to 0.24, right? Anyway, I updated to 0.24 and it worked finally, thanks!

hoechenberger, 2021-11-13:
Great to hear it works now!! 👍