mne-tools/mne-python #8850: [DOC] Add `Brain._renderer.set_camera` to `stc.plot` documentation

alexrockhill, 2021-02-09:
I noticed that it wasn't described in the `SourceEstimate.plot` documentation how to set a view that is not one of the canonical views. I had to look in the source code to remember how to do this so I thought it might be useful to document. It does expose a private function as in `brain._renderer.set_camera(azimuth=-110, elevation=100, distance=500, np.array([-0.03326882, -0.04459817,  0.08267832]))`. This is quite messy/showing the under the hood bits that maybe shouldn't be shown but I'm not sure how else to get the view you really like without the GUI.

alexrockhill, 2021-02-09:
If this is not helpful, no worries.

GuillaumeFavelier, 2021-02-09:
Are you aware of the public function called `show_view()` ? It uses `set_camera()` under the hood and accepts `elevation`, `azimuth`, `distance` ...etc.
Let me know if there is something that you can't accomplish with it.

alexrockhill, 2021-02-09:
Ah no, that is a much better solution. Maybe there is a better way to make this easier to find from the documentation or maybe I should have just looked harder :)

GuillaumeFavelier, 2021-02-09:
No worries, the `Brain`'s API is a work in progress so any suggestion of improvement is welcome.

> Maybe there is a better way to make this easier to find from the documentation

I don't know how though 😅

alexrockhill, 2021-02-09:
Maybe a first step would be changing the `SourceEstimate.plot` returns to `mne.viz.Brain` from `surfer.Brain` as that was what was returned for me. That would be a good step in pointing users in the right direction unless I'm wrong about the return object :)

GuillaumeFavelier, 2021-02-09:
That is a good point! 👌

PR #8851 (2021-02-09): DOC: plot_stc
This PR hopefully helps clarify the return value of `plot_stc` as described in https://github.com/mne-tools/mne-python/issues/8850#issuecomment-776180591.

Closes #8850
closed 2021-02-09 by hoechenberger