mne-tools/mne-python #8853: Confusion around "resp" channel type

hoechenberger, 2021-02-10:
In `channels.py`, it is claimed that the `resp` channel type is for certain response triggers:

https://github.com/mne-tools/mne-python/blob/19fb479de44865d5ce79ad5af539a06081fae0c5/mne/channels/channels.py#L704-L706

This seems to be supported by code treating BTi data:
https://github.com/mne-tools/mne-python/blob/19fb479de44865d5ce79ad5af539a06081fae0c5/mne/io/bti/bti.py#L1188-L1189

But the following comment in `constants.py` suggests it's a resp-iratory, not a resp-onse channel:
https://github.com/mne-tools/mne-python/blob/19fb479de44865d5ce79ad5af539a06081fae0c5/mne/io/constants.py#L177

Now … what gives?

x-ref https://github.com/mne-tools/mne-bids/issues/700#issuecomment-776632491

cc @agramfort

larsoner, 2021-02-10:
The authority should be fiff-constants

https://github.com/mne-tools/fiff-constants/blob/master/DictionaryTypes.txt#L257

hoechenberger, 2021-02-10:
So… it's our BTi code that needs fixing, not the comment in `constants.py`?

larsoner, 2021-02-10:
Yes I think so

sappelhoff, 2021-02-10:
awesome, thanks for digging this out @hoechenberger ... this is good for MNE-BIDS :)

PR #8856 (2021-02-10): "resp" channel is for respiration, not responses
Fixes #8853
commit 2c6b0dd6 (2021-02-10): "resp" channel is for respiration, not responses (#8856)

* "resp" channel is for respiration, not responses

Fixes #8853

* review suggestions
closed 2021-02-10 by hoechenberger