mne-tools/mne-python #8855: GSoC idea

cbrnr, 2021-02-10:
I've been thinking that a fast 2D signal browser would be nice to have. Would it be OK to add a project to our GSoC ideas? Here's a draft, let me know what you think:

## 6. Write high-performance 2D visualization for browsing signals
### Difficulty
Medium (requires experience with GUI toolkits such as Qt/PyQt)

### Possible mentors
[Clemens Brunner](https://github.com/cbrnr)

### Goal
MNE-Python currently offers a Matplotlib-based signal browser to visualize EEG/MEG time courses (`raw.plot()`). Although this browser gets the job done, it is rather slow (smooth scrolling is not possible) and uses custom (i.e. non-standard) GUI elements. The goal of this project is to implement an alternative signal browser that solves these issues, i.e. it should support smooth scrolling through the data and use standard GUI elements. The browser should also support working with annotations. If possible, Qt-based solutions should be preferred, and of course the result should be written in pure Python and work across platforms. Therefore, potential solutions include using [PyQtGraph](http://www.pyqtgraph.org/) or writing custom Qt widgets.

### Subgoals
- Determine optimal toolkit/package to implement a high-performance browser
- Implement signal browser
- Write CI tests

agramfort, 2021-02-10:
I've been thinking about this too. I would start by something in a
satellite repo of mne-python
which is also easier for new comers that do not start from a huge legacy
code base

>

cbrnr, 2021-02-10:
Yes, this can be a relatively independent project, and once it's ready it can be easily integrated. But do you agree that it should still be part of MNE-Python GSoC? A satellite repository wouldn't qualify for GSoC I think.

drammock, 2021-02-10:
This fits with our discussion last week at office hours about moving UI controls out of the brain viewer figure window and into the toolbar.  I agree with movement in that direction for 2D plots as well.  My only hesitation is that I think getting the raw/epochs/ICA data browsers to be properly interactive in a notebook is an equally high (or maybe higher) priority...  so in a perfect world I would want whatever solution gets implemented here to be compatible with a solution to the notebook embedding problem.

cbrnr, 2021-02-10:
Yes, I haven't thought about notebooks at all. Personally, I don't use them and I don't think it's worth supporting interactive data exploration in a notebook, but maybe that's just me. In any case, I guess this would invalidate all Qt-based solutions I know and it's probably better to look into web-based viz tools.

drammock, 2021-02-10:
> Personally, I don't use them and I don't think it's worth supporting interactive data exploration in a notebook, but maybe that's just me.

I don't use them often either, but it seems a lot of users do.  I think we need to support both.  There are also users trying to embed raw data browser into other applications (e.g., https://github.com/mne-tools/mne-python/issues/8142).  What I'm envisioning is a design where we have some kind of data object with an "upper" API for taking input from qt widgets, ipywidgets, or whatever, and a "lower" API for the backend canvas (matplotlib, pyqtgraph, whatever).  I'm hoping that it's possible to build out a PyQtGraph-based solution for a fast desktop data browser that gets us part of the way toward that goal / makes interactive (if slow) data browsers in the notebook (or embedded in other apps) easier to achieve later.

cbrnr, 2021-02-10:
This sounds like a good idea. So I'd focus on a specific backend in this project then. I have a basic concept working with PyQtGraph, and I've also started to implement something using base Qt. I think the latter will be much more customizable to do what we really want, but of course more work.

cbrnr, 2021-02-10:
I've updated the project description, feel free to edit @drammock.

drammock, 2021-02-10:
I've edited the last paragraph. Thanks @cbrnr

cbrnr, 2021-02-12:
Thanks @drammock, I'll add that to the wiki.

closed 2021-02-12 by cbrnr

larsoner, 2021-02-18:
10 sec of 376 channels scrolls pretty smoothly on my machine using VTK using [this gist](https://gist.github.com/larsoner/7acaeb67f79586975d0b07584a8fc5c9) (after applying the necessary fix from a related [VTK issue](https://gitlab.kitware.com/vtk/vtk/-/issues/18122#note_904930)), which we've gotten to embed pretty nicely lately in notebooks and Qt (here Qt):

https://user-images.githubusercontent.com/2365790/108407187-6629bb00-71f1-11eb-9eee-c493b5b51678.mp4

larsoner, 2021-02-18:
Sorry that was actually only 2 sec of data, at 10 sec it is a little bit slower:

https://user-images.githubusercontent.com/2365790/108407536-d9333180-71f1-11eb-83ba-695a6869209c.mp4

But still I think performance is good enough already here, and I haven't tried to optimize the data upload/replacement or anything.

cbrnr, 2021-02-18:
This looks promising, but the key issue will be how to display a large data set at various scales. I guess these issues are independent from the frontends (VTK, Qt), but important for smooth visualization. I haven't played around yet, but I'd start with keeping only the visible segment in memory and loading subsequent segments on demand from disk. Of course this gets trickier with large time scales, where 1 pixel corresponds to e.g. 1000 samples or more. I guess downsampling will be necessary, but we'd have to test if this if fast enough on demand.

larsoner, 2021-02-18:
> I haven't played around yet, but I'd start with keeping only the visible segment in memory and loading subsequent segments on demand from disk.

Based on trying to optimize our matplotlib backend a lot (@drammock let me know if you agree) the drawing is the biggest bottleneck. Our raw data reads are not (currently at least) an appreciable problem relative to the draw time. So to me the fact that VTK can display over 2 million data points smoothly above suggests that it will alleviate our biggest bottleneck.

To test if anything other than drawing will be a bottleneck, you can take our raw plotting code and comment out any `draw` calls. Then try interacting and see if it's snappy enough. My guess is that for most use cases it will be.

cbrnr, 2021-02-18:
Are you sure it's drawing all data points? Or maybe it intelligently handles drawing such a large amount of data. In my naive tests with Qt, drawing got slower and slower the larger the data was, even though the same amount of points were visible at a time.

larsoner, 2021-02-18:
> Are you sure it's drawing all data points? Or maybe it intelligently handles drawing such a large amount of data. In my naive tests with Qt, drawing got slower and slower the larger the data was, even though the same amount of points were visible at a time.

I'm not quite sure what you have in mind here, but I'm pretty sure it would just redraw all shown data points every refresh (no caching / blitting / etc.). In the code I actually delete the data table and recreate it from scratch based on the time slice needed for each draw, so VTK wouldn't know if the data was the same or not without doing something like a hash, and even in that case I'm not replacing with identical values (only some rows are the same, but they are even shifted) so their logic would have to be insanely complicated to know what they do and don't need to draw and how to blend them properly.

But just to make absolutely sure it's drawing all 2.2m points of the displayed data slice on each draw I now update the data to be shown with a `rand()`-scaled version of it, and it draws at the same speed as before:

https://user-images.githubusercontent.com/2365790/108414387-2adfba00-71fa-11eb-8d56-e7d686f2b4ab.mp4

cbrnr, 2021-02-18:
OK, so you're just giving it the data that's currently visible? I think it might be the way I'm currently doing it with QGraphicsView, I put *all* data into the scene and put the view on the currently visible part. Strangely, more data in the scene slows down drawing of the view. I might just be doing it wrong though, need to look into it but currently I don't have a lot of time.

Can you modify your demo to scroll with the arrow keys instead of automatically?

larsoner, 2021-02-18:
No this is just POC territory for me but you're welcome to try it

cbrnr, 2021-02-19:
New lib on the block: https://datoviz.org/ (successor to VisPy)

larsoner, 2021-02-19:
AFAIK Vulkan is cutting-edge viz, I'm not sure how well supported it will be even in the next couple of years. The warning "but with up-to-date graphics drivers" is an important one.

There is also https://github.com/pygfx/wgpu-py based on https://github.com/gfx-rs/wgpu which could be another way forward. It abstracts over Vulkan/Metal/GL which should greatly help compatibility, and is somehow related to Mozilla (probably has some nice backing).

You could try mocking something in either or both of these to see how difficult it is to get working, see if picking and zooming and axis labeling works, etc.

cbrnr, 2021-02-19:
> AFAIK Vulkan is cutting-edge viz, I'm not sure how well supported it will be even in the next couple of years. The warning "but with up-to-date graphics drivers" is an important one.

Yes, although I was always under the impression that OpenGL is so old that "up-to-date graphics drivers" should apply to any reasonably modern computer including my 2014 MacBook Pro (which many people would call old).

WebGPU looks nice indeed. It would be nice to have some minimal PoC in both frameworks, but it seems like WebGPU has a much lower entry barrier because it's pip installable (whereas everything needs to be built for Datoviz at the moment).

I don't know, I guess I'll try optimizing what I have with PyQt before trying out other stuff. At least PyQt has been around for years and is likely to stick around for many years to come.

larsoner, 2021-02-19:
> I don't know, I guess I'll try optimizing what I have with PyQt before trying out other stuff. At least PyQt has been around for years and is likely to stick around for many years to come.

This is part of why VTK appeals to me. It's not going anywhere and is reasonably well supported. We're also already using VTK for 3D viz so we're not really adding any maintenance overhead if we start using it for other things -- we're already tied to it for some stuff.