mne-tools/mne-python #8859: SPM faces

mirfaridmusavian, 2021-02-11:
The eeg signals have not been included in the mne.datasets.spm_face.data_path()

larsoner, 2021-02-11:
Indeed we should add them at some point. @mirfaridmusavian would you be up for improving the SPM faces example to use the EEG data?

agramfort, 2021-02-11:
you would have to start from the files:

faces_run1.bdf
faces_run2.bdf
electrode_locations_and_headshape.sfp

Alex

mirfaridmusavian, 2021-02-12:
I can extract the epochs with the SPM12 toolbox in matlab and share, would it be enough?

larsoner, 2021-02-12:
I think a better solution would be to start from the original BDF files as suggested by @agramfort. Really we want to show how to go from raw data to finished analysis using MNE-Python whenever possible

mirfaridmusavian, 2021-02-15:
I can give it a try on coming weekend