mne-tools/mne-python #8864: ValueError: Could not find EEG array in the .set file.

MSA8D8, 2021-02-12:
#### Describe the bug
Dear MNE developers
Thank you for developing this wonderful tool. 

I am faced with the error when I tried to use read_raw_eeglab() to load the .set file made by EEGLAB
What I did in EEGLAB is just loading the data by Quick 30 and save it in '.set(with '.fdt')' format only. 
I think this is very weird, so I guess this is bug.

The script I made is below.
```
subject_name = 'DA'
script_path = os.path.abspath('./Python/EEG_analysis')
data_path = os.path.abspath('./output_play/' + subject_name) 
os.path.exists(data_path + '/'+ subject_name + '.set') # The print out result is 'True'
raw = mne.io.read_raw_eeglab(input_fname = data_path + '/'+ subject_name + '.set')
```
#### Actual results
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-17-1519c0d84577> in <module>
----> 1 raw = mne.io.read_epochs_eeglab(input_fname = data_path + '/'+ subject_name + '_play_merged.set')

~/.pyenv/versions/anaconda3-5.3.1/lib/python3.7/site-packages/mne/io/eeglab/eeglab.py in read_epochs_eeglab(input_fname, events, event_id, eog, verbose, uint16_codec)
    277     epochs = EpochsEEGLAB(input_fname=input_fname, events=events, eog=eog,
    278                           event_id=event_id, verbose=verbose,
--> 279                           uint16_codec=uint16_codec)
    280     return epochs
    281 

<decorator-gen-198> in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, montage, eog, verbose, uint16_codec)

~/.pyenv/versions/anaconda3-5.3.1/lib/python3.7/site-packages/mne/io/eeglab/eeglab.py in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, montage, eog, verbose, uint16_codec)
    449                  uint16_codec=None):  # noqa: D102
    450         eeg = _check_load_mat(input_fname, uint16_codec)
--> 451 
    452         if not ((events is None and event_id is None) or
    453                 (events is not None and event_id is not None)):

~/.pyenv/versions/anaconda3-5.3.1/lib/python3.7/site-packages/mne/io/eeglab/eeglab.py in _check_load_mat(fname, uint16_codec)
     63             'mne-python developers for more information.')
     64     if 'EEG' not in eeg:
---> 65         raise ValueError('Could not find EEG array in the .set file.')
     66     else:
     67         eeg = eeg['EEG']

ValueError: Could not find EEG array in the .set file.
```

#### Additional information
I am using the latest version of EEGLAB(2021), so there might be some minor change in the .set file. This might be why I am faced with this error.

Thank you in advance.

cbrnr, 2021-02-12:
Could you share a small example file?

MSA8D8, 2021-02-12:
@cbrnr Thank you for your quick reply.
Here is the link for the sample data. 
https://drive.google.com/drive/folders/1UQ9vP_-FZ-ZAipH8KroGZ3Rhs5IpybNw?usp=sharing

I will give you the access permission of the link when you download them.

I made this sample in the same way as before (just loading the data by Quick 30 and save it in '.set(with '.fdt')' format only).
The EEGLAB version is 2021.0

Thank you for your help.

cbrnr, 2021-02-12:
Usually, the `.set` file (which is just a `.mat` file) contains an `EEG` key with all relevant data. Your file does not have this `EEG` key, but all fields are directly in the root of the `.mat` file.

Can you please double-check that you can load this file with EEGLAB? If they changed the format we need to adapt our reader, but I want to make sure that this is a valid EEG file.

cbrnr, 2021-02-12:
@MSA8D8 can you please also post the output of `eeghist` so that we can see how you exported the data from EEGLAB?

MSA8D8, 2021-02-12:
@cbrnr Thank you for your quick reply.

> Can you please double-check that you can load this file with EEGLAB?

Here is the Matlab output of loading the sample data and checking the history of it. 
I am sure that this is a valid EEG file. 

Thank you for your help.

```
>> eeglab
Some menus items hidden. Use Preference menu to show them all.
eeglab: options file is ~/eeg_options.m
Retrieving plugin versions from server...
Retreiving download statistics...
EEGLAB: adding "Fieldtrip-lite" to the path; subfolders (if any) might be missing from the path
EEGLAB: adding "Fileio" to the path; subfolders (if any) might be missing from the path
EEGLAB: adding "ICLabel" v1.3 (see >> help eegplugin_iclabel)
EEGLAB: adding "Viewprops" v1.5.4 (see >> help eegplugin_viewprops)
EEGLAB: adding "bva-io" v1.7 (see >> help eegplugin_bva_io)
EEGLAB: adding "clean_rawdata" v2.3 (see >> help eegplugin_clean_rawdata)
EEGLAB: adding "dipfit" v3.7 (see >> help eegplugin_dipfit)
EEGLAB: adding "firfilt" v2.4 (see >> help eegplugin_firfilt)
You are using the latest version of EEGLAB.
pop_loadset(): loading file /Users/masarusasaki/Documents/GitHub/Script/demo/Sample.set ...
Reading float file '/Users/masarusasaki/Documents/GitHub/Script/demo/Sample.fdt'...
Creating a new ALLEEG dataset 1
Done.
>> EEG

EEG = 

  フィールドをもつ struct:

             setname: 'Sample'
            filename: 'Sample.set'
            filepath: '/Users/masarusasaki/Documents/GitHub/Script/demo/'
             subject: ''
               group: ''
           condition: ''
             session: []
            comments: 'Original file: sasakitest20200610arest.eeg'
              nbchan: 29
              trials: 1
                pnts: 154624
               srate: 500
                xmin: 0
                xmax: 309.2460
               times: [1×154624 double]
                data: [29×154624 single]
              icaact: []
             icawinv: []
           icasphere: []
          icaweights: []
         icachansind: []
            chanlocs: [1×29 struct]
          urchanlocs: []
            chaninfo: [1×1 struct]
                 ref: 'common'
               event: []
             urevent: []
    eventdescription: {}
               epoch: []
    epochdescription: {}
              reject: [1×1 struct]
               stats: [1×1 struct]
            specdata: []
          specicaact: []
          splinefile: ''
       icasplinefile: ''
              dipfit: []
             history: '↵EEG.etc.eeglabvers = '2021.0'; % this tracks which version of EEGLAB is being used, you may ignore it↵EEG = pop_loadbv('/Users/masarusasaki/Documents/GitHub/Script/demo/sasakidata/raw/2020-6-10/', 'sasakitest20200610arest.vhdr', [1 154624], [1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29]);↵EEG.setname='Sample';↵EEG = eeg_checkset( EEG );↵EEG = eeg_checkset( EEG );'
               saved: 'yes'
                 etc: [1×1 struct]
                 run: []
             datfile: 'Sample.fdt'
```

cbrnr, 2021-02-12:
@agramfort @hoechenberger @mmagnuski the fix is relatively simple (instead of raising an error [here](https://github.com/mne-tools/mne-python/blob/f70f508a1bd5f73bdb4797e657340c11cc138d64/mne/io/eeglab/eeglab.py#L65) just continue because we already have all the fields that are otherwise contained within `EEG`), but before fixing this I'd like to hear what you think.

hoechenberger, 2021-02-12:
Seems this is a change that was introduced with EEGLAB 2021:
https://eeglab.org/others/EEGLAB_revision_history.html#eeglab-version-20210

I wonder if we could explicitly check for the existence of all requires fields before proceeding?

cbrnr, 2021-02-12:
Hmm, where do you see this change? It says "Now saving all EEGLAB datasets in a single file instead of two (fully backward compatible)", but it doesn't mention dropping the `EEG` struct. Also, the example data set still consists of two files.

If we're sure this is a (new) valid format I'd not bother and check all fields, for one because I haven't found any official description of the format. Also, if a required field is missing it will fail at some point anyway.

hoechenberger, 2021-02-12:
> Hmm, where do you see this change? It says "Now saving all EEGLAB datasets in a single file instead of two (fully backward compatible)", but it doesn't mention dropping the `EEG` struct.

Yeah I just assumed that since they changed something, and since @MSA8D8's dataset was created using EEGLAB 2021, this could be related.

cbrnr, 2021-02-15:
@arnodelorme sorry to bother you again, but could you confirm if EEGLAB now saves .set/.fdt data fields in the root instead of within an `EEG` struct? All of our test files contain the `EEG` struct, but @MSA8D8 has data (created with the latest EEGLAB version) where this is not the case.

arnodelorme, 2021-02-15:
This is correct.

arnodelorme, 2021-02-15:
This is because, when saving in the root, it is possible to ignore some fields (such as "data"). In that case, we can load the metadata without loading the data, so there is no need to save 2 files anymore.

cbrnr, 2021-02-15:
Thanks @arnodelorme! I'll prepare a fix for our reader!

jeangillain, 2021-03-03:
Hello, 

I was wondering if you fixed the problem, because when I try to load .set files on mne, I still have the same problem. 
Thanks a lot for everything you do,

Jean Gillain

hoechenberger, 2021-03-03:
Hello @jeangillain,

this has been fixed in our development and 0.22 maintenance branches; but there hasn't been a official release since this fix was added. You can install the latest version of the (unreleased) 0.22 maintenance branch (which will eventually become 0.22.1) by running:

```shell
pip install -U https://github.com/mne-tools/mne-python/archive/maint/0.22.zip
```
I hope this helps!

MSA8D8, 2021-03-30:
Hello. @cbrnr @hoechenberger 

I still have this problem even after installing the latest version of mne as @hoechenberger suggested.

The version of mne I can install is 0.22.0

Thank you in advance.

cbrnr, 2021-03-30:
Can you share a small example file again? I can't find the link you posted previously.

MSA8D8, 2021-03-30:
@cbrnr Thank you for your help.
Here is the link for the same data I shared with you before.
https://drive.google.com/drive/folders/1rQAyEMYBxiBgNHQSbphIsb_spFmmXwY-?usp=sharing

cbrnr, 2021-03-30:
It works for me with the current maint branch (linked above). Maybe you still have the 0.22 release? Unfortunately, the version shows 0.22 in both the latest release and the WIP maint branch, so I don't know how to determine if you have installed it correctly.

In any case, I think it's time for an official release of 0.22.1 - @hoechenberger @larsoner what are our plans?

agramfort, 2021-03-30:
>
> In any case, I think it's time for an official release of 0.22.1 -
> @hoechenberger <https://github.com/hoechenberger> @larsoner
> <https://github.com/larsoner> what are our plans?
>

+1

hoechenberger, 2021-03-30:
Yes +10 from me for 0.22.1 sooner than later!!

MSA8D8, 2021-03-30:
@cbrnr 
Thank you for checking the problem.

Here is the things when I tried to upgrade the version of mne.

```
MacBook-Pro ~ % pip install -U https://github.com/mne-tools/mne-python/archive/maint/0.22.zip
Collecting https://github.com/mne-tools/mne-python/archive/maint/0.22.zip
  Using cached https://github.com/mne-tools/mne-python/archive/maint/0.22.zip
Requirement already satisfied: numpy>=1.11.3 in ./.pyenv/versions/anaconda3-5.3.1/lib/python3.7/site-packages (from mne==0.22.0) (1.19.5)
Requirement already satisfied: scipy>=0.17.1 in ./.pyenv/versions/anaconda3-5.3.1/lib/python3.7/site-packages (from mne==0.22.0) (1.6.0)
WARNING: You are using pip version 21.0; however, version 21.0.1 is available.
You should consider upgrading via the '/Users/masarusasaki/.pyenv/versions/anaconda3-5.3.1/bin/python -m pip install --upgrade pip' command.
MacBook-Pro ~ % 

```

After doing this, I am still faced with the same error.

MSA8D8, 2021-03-30:
@cbrnr 
I uninstalled mne and install it again as @hoechenberger suggested.
Then, the problem was finally solved !!!

Thank you for everything !!
I really appriciate all of your kindly help !!

PR #8874 (2021-02-17): Fix EEGLAB import
EEGLAB files may contain all fields in the root of the file instead of within an `EEG` struct. This PR adds support for this format. Fixes #8864.
closed 2021-03-30 by MSA8D8