mne-tools/mne-python #8865: mne.read_selection() returning funny channels

dnacombo, 2021-02-12:
#### Describe the bug

trying to select channels from an EEG dataset.

#### Steps to reproduce

```
l = ['Vertex', 'temporal', 'parietal', 'occipital', 'frontal']

ll = mne.read_selection(name=l, info=r.info)
```
returns a list with **Neuromag** channels although there are only EEG channels (EGI) in the dataset.

@hoechenberger

hoechenberger, 2021-02-12:
`mne.read_selection()` shouldn't return channels that don't appear in `info`, so I'd consider this a bug

larsoner, 2021-02-12:
IIRC `read_selection` is only meant to be used with Neuromag MEG channels. We should probably also update the docstring to say this in addition to raising an error

cbrnr, 2021-02-12:
👍 - I've never heard of that function, and after reading its docs I couldn't figure out what exactly it should be used for.

agramfort, 2021-02-12:
I was written in a time where MNE could only process neuromag data....

>

drammock, 2021-02-12:
Maybe worth a deprecation and a name change to `read_neuromag_selection`?

cbrnr, 2021-02-12:
👍 for deprecating and renaming, and maybe also moving to the `channels` module?

hoechenberger, 2021-02-12:
I will look into this.

PR #8870 (2021-02-18): MRG: selection.read_selection() ➡️ channels.read_vectorview_selection()
Fixes #8865

@larsoner There is a bug in `deprecated_alias`:
```python
/Users/hoechenberger/Development/mne-python/mne/utils/docs.py:2551: DeprecationWarning: Function 
read_neuromag_selection is deprecated; read_selection has been deprecated in favor of
read_neuromag_selection and will be removed in 0.24
```
In the sentence before the semicolon, it refers to the new function name, not to the deprecated one.

#### Additional information
I moved the API section of the changelog up above the bugs section, and added a new sec [... cut]
commit 53f1b859 (2021-02-18): MRG: selection.read_selection() ➡️ channels.read_vectorview_selection() (#8870)

* Move read_selection() to channels.read_neuromag_selection()

Fixes #8865

* Fix Sphinx warning

* read_neuromag_selection -> read_vectorview_selection

* Use pytest context manager

* Add API reference

* Fix changelog

* Update docstring

* Update mne/channels/channels.py

* Docstring, import

* fix changelog [skip azp][skip github]
closed 2021-02-18 by larsoner