mne-tools/mne-python #8866: wrong scale for patterns with SGDClassifier

giuliagennari, 2021-02-12:
#### Describe the bug
When using a SGDClassifier, the patterns computed with LinearModel/get_coef come out with a fanny scale i.e. in the wrong order of magnitude.

#### Steps to reproduce
```python
# these lines come from the online tutorial, I've just substituted the default logistic regression 
# with a logistic regression implemented by means of the SGDClassifier 

import mne
from mne import io, EvokedArray
from mne.datasets import sample
from mne.decoding import Vectorizer, LinearModel, get_coef

from sklearn.preprocessing import StandardScaler
from sklearn.linear_model import SGDClassifier
from sklearn.pipeline import make_pipeline

data_path = sample.data_path()
sample_path = data_path + '/MEG/sample'

raw_fname = sample_path + '/sample_audvis_filt-0-40_raw.fif'
event_fname = sample_path + '/sample_audvis_filt-0-40_raw-eve.fif'
tmin, tmax = -0.1, 0.4
event_id = dict(aud_l=1, vis_l=3)

# Setup for reading the raw data
raw = io.read_raw_fif(raw_fname, preload=True)
raw.filter(.5, 25, fir_design='firwin')
events = mne.read_events(event_fname)

# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    decim=2, baseline=None, preload=True)
del raw

labels = epochs.events[:, -1]

X = epochs.pick_types(meg=False, eeg=True)
y = epochs.events[:, 2]

clf = make_pipeline(
    Vectorizer(),                       # 1) vectorize across time and channels
    StandardScaler(),                   # 2) normalize features across trials
    LinearModel(SGDClassifier(loss='log')))  # 3) fits a logistic regression with Stochastic Gradient Descent 
clf.fit(X, y)

# Extract and plot patterns and filters
for name in ('patterns_', 'filters_'):
    # The `inverse_transform` parameter will call this method on any estimator
    # contained in the pipeline, in reverse order.
    coef = get_coef(clf, name, inverse_transform=True)
    evoked = EvokedArray(coef, epochs.info, tmin=epochs.tmin)
    evoked.plot_topomap(title='EEG %s' % name[:-1], time_unit='s')
```

#### Additional comments

Perhaps not a real bug! If so.... would you know how come???

agramfort, 2021-02-12:
how many samples do you have? using SGDClassifier is adapted with lots of
samples

>

giuliagennari, 2021-02-12:
I am training my SGDclassifiers on ~200 epochs/subject. I am using this classifier to exploit the possibility of .partial_fit() . 
I first validated my choice by finding that with a high-enough number of iterations SGDClassifier(loss='log') produces almost identical outcomes as LogisticRegression()

agramfort, 2021-02-12:
what are n_samples and n_features for each classifier for you? why is
partial_fit required? data
are really too big to fit in RAM?

>

giuliagennari, 2021-02-12:
my n_samples is ~200 with 1250 features. It is not a matter of RAM or anything related to computational resources, my aim is to train the _same_ classifiers on _distinct_ training sets while keeping the latter separate. This strategy yields, for my particular design in a particular circumstance, a cleaner performance relative to just training distinct classifiers on the distinct training sets and then averaging their final scores. Because of my design it is important to keep the training sets separate.

However, I think all these facts are unrelated to the patterns.... If you run the code above you'll see that the patterns seems correct beside the scale which does not make any sense

agramfort, 2021-02-13:
have you checked that the regularization strength and the scaling of the
input features is the same for both estimators?

>

giuliagennari, 2021-02-13:
The input features are the same yes. 
I always use L2-penalty if that is what you are asking. Otherwise I do not know what regularization strength is!

agramfort, 2021-02-13:
alpha is not 1 by default in SGD. Use

```
 LinearModel(SGDClassifier(alpha=1., loss='log'))  # 3) fits a logistic regression with Stochastic Gradient Descent
```

and it works as you expected.

closed 2021-02-13 by agramfort

giuliagennari, 2021-02-14:
All right. I am grateful for your assistance (...and in the week-end!) **_Thank you_**