mne-tools/mne-python #8868: Inaccurate iEEG electrode positions on 3d/2d plots

Immiora, 2021-02-12:
Originally posted here: https://mne.discourse.group/t/ieeg-electrode-positions-on-3d-2d-plots/2687

The electrode positions I am getting from the mayavi plot look inaccurate.

The issue is well visible in the tutorial as well: [Working with ECoG data](https://mne.tools/dev/auto_tutorials/misc/plot_ecog.html#sphx-glr-auto-tutorials-misc-plot-ecog-py):
[image]

The electrodes plotted with `scatter` (colored blobs) are off compared to the ones made with `plot_alignment` (white spheres underneath). Reducing the radius of the scatter plot reveals how inaccurate it is:

[image]

I understand that these positions depend on the view, but both plots are made using the same viewpoint:

```
fig = plot_alignment(raw_car.info, subject=subject, subjects_dir=subjects_dir, surfaces=['pial'], coord_frame='mri')
mne.viz.set_3d_view(fig, 200, 70)
xy, im = snapshot_brain_montage(fig, montage)
xy_pts = np.vstack([xy[ch] for ch in raw_car.info['ch_names']])

fig, ax = plt.subplots(figsize=(5, 5))
ax.imshow(im)
ax.set_axis_off()
sc = ax.scatter(*xy_pts.T, c=ts, vmin=10, vmax=100)
```

Immiora, 2021-03-02:
Any chance someone is looking at this yet?

larsoner, 2021-03-03:
Can you try #8983?

PR #8983 (2021-03-03): MRG, FIX: Fix coordinate calculations
Closes #8868

Love that VTK provides an even simpler solution to this problem!
closed 2021-03-03 by larsoner