mne-tools/mne-python #8871: Improve Epochs docs (and more)

cbrnr, 2021-02-15:
Based on my question in the [forum](https://mne.discourse.group/t/computing-tfr-drops-bad-epochs/2691), I suggest to clarify the `preload` parameter to make it clear that preloading does *not* depend on the status of `raw.preload` (which I assumed was the case).

I'd change the `preload` description as follows:

> Load all epochs ~~from disk~~ into memory when creating the object or wait before accessing each epoch (more memory efficient but can be slower).

In addition, I'd add the following note:

> Epochs are preloaded based on the value of the `preload` argument. Even if the `raw` argument has already been preloaded, the returned `Epochs` object will **not** be preloaded by default. This means that bad epochs (e.g. epochs with not enough data) will not be dropped immediately, so the number of good epochs displayed may change once the epochs are preloaded.

@hoechenberger @larsoner @agramfort @drammock since this is such a widely used object I'd be grateful if you commented and edited my suggestion if necessary.

Furthermore, I noticed that `preload` is used inconsistently across `BaseEpochs`, `Epochs`, `EpochsArray`, and `EpochsFIF` classes:

- `BaseEpochs`: Doesn't have a `preload` parameter but `preload_at_end` - what's the reason to use a different parameter?
- `Epochs`: `preload=False`
- `EpochsArray`: Has neither `preload` nor `preload_at_end` parameters.
- `EpochsFIF`: `preload=True`

I think it would be great to double-check these inconsistencies - there are probably good reasons, but if not we should discuss changing accordingly.

Finally, @hoechenberger suggested to discuss `preload=None` as a default, meaning that this would re-use its `raw.preload` value.

drammock, 2021-02-15:
Proposed new docstring:

> Whether to load the epochs into RAM when creating the object, or to access the epochs from disk when needed (more memory efficient but can be slower). Note that some operations require epochs to be loaded into RAM (e.g., epoch rejection based on peak values, setting the EEG reference, etc). Loading into RAM can also be done later with `mne.Epochs.load_data`. Default is ``False``.

cbrnr, 2021-02-15:
Thanks @drammock. I assume that this description does not need an additional note anymore.

Also, I haven't checked the implementation, but is it really true that `preload=False` always implies loading from disk? I assumed that if I pass a preloaded `Raw` object to `Epochs(..., preload=False)`, the data used to create epochs is accessed from RAM (the preloaded `Raw` object) instead of going back to loading from disk. What if a `Raw` object isn't associated with a file?

cbrnr, 2021-02-15:
Furthermore, the gotcha I ran into doesn't have anything to do with "some operations", which sounds as if this might happen only in some cases. Eventually, the `Epochs` objects has to be created, and that's where epoch rejection automatically takes place.

drammock, 2021-02-15:
> Also, I haven't checked the implementation, but is it really true that `preload=False` always implies loading from disk?

it's easy enough to test the behavior just by preloading a raw, then epoching it with `preload=False` and a rejection dict specified.  Indeed, the epoch rejection won't occur upon object creation; only after explicitly calling `epochs.load_data()`.  You're probably right that strictly speaking this does not mean the epochs are "on disk"; I'd guess they store a reference to the raw object that prevents it getting GCd.  I think you'd have to read the source to be sure.

cbrnr, 2021-02-15:
I just checked, and it seems like it really does load epochs from disk even if the `Raw` object is already loaded into RAM. This seems a bit inefficient, but I guess this is not what this issue is about. Let's focus on improving the docstring and maybe making the `preload` business a bit more consistent.

agramfort, 2021-02-15:
let me try to bring some clarification. The motivation was that by default
creating an Epochs object
was immediate. So even if raw.preload=True then no rejection is done and
nothing is allocated.
Also you can drop bad epochs without keeping everything is RAM. It will
read eventually drop
bad epochs and then release the memory. See calling epochs.drop_bad()  on
epochs with
preload=False.

>

cbrnr, 2021-02-17:
Thanks @agramfort, I think this makes sense, but it isn't very clear. So before I open a PR, can we discuss the inconsistent usage of `preload` parameters in the four classes (see [top comment](https://github.com/mne-tools/mne-python/issues/8871#issue-808432779))?

larsoner, 2021-02-17:
EpochsFIF has `preload=True` by default for historical reasons. It seems overkill to me to change-by-deprecation this to `preload=False` but we could do it.

Other than this, to me everything is consistent and we should just make clear that preload means "should this instance load all of its data into memory". Then Raw having `preload=True` (or `False`) means the given instance hold all of its data in memory (or not), Epochs having `preload=True` (or `False`) means its instance holds all of its data in memory (or not). From this it seems clear why doing `read_raw_fif(..., preload=True)` followed by `Epochs(..., preload=False)` would mean the Raw instance has all of its data in memory but the Epochs instance does not.

cbrnr, 2021-02-17:
> `EpochsFIF` has `preload=True` by default for historical reasons. It seems overkill to me to change-by-deprecation this to `preload=False` but we could do it.

I'd do it.

The way you put it makes the reasoning very clear, I'll adopt something like this somewhere in the docstring.

However, what about

- `BaseEpochs`: Doesn't have a `preload` parameter but `preload_at_end` - what's the reason to use a different parameter?
- `EpochsArray`: Has neither `preload` nor `preload_at_end` parameters.

larsoner, 2021-02-17:
> BaseEpochs: Doesn't have a preload parameter but preload_at_end - what's the reason to use a different parameter?

No users should construct use this class on their own, so I'm not worried about this. (If it's not clear that users shouldn't use this, we should add a `.. warning::` to the docstring at the top.) To know why you'd have to dig into the code, I don't remember offhand.

> EpochsArray: Has neither preload nor preload_at_end parameters.

The data are already in an array, i.e., in memory.

cbrnr, 2021-02-17:
Re `BaseEpochs`, it basically uses `preload_at_end` like `preload` (or rather, we could easily change it to `preload`).

Re `EpochsArray`, the same could be said about preloaded `raw` (epochs just slice out the relevant data).

larsoner, 2021-02-17:
> we could easily change it to preload

My experience with making changes to `Epochs` is that anything involving data loading is never easy and has unintended consequences. If you want to try to work through it I guess we could do it, but I think there are probably many targets more worthy of our attention in our codebase.

> Re EpochsArray, the same could be said about preloaded raw (epochs just slice out the relevant data).

The key part here is the "epochs just slice out the relevant data" -- in the `preload=False` case this happens on demand one epoch at a time as they are accessed and these data are not kept/stored in memory by the epochs instance. In the `preload=True` case, all these data slices are stored together (and held in a new, potentially very memory-consuming `ndarray` in memory separate from that of the raw instance) by the epochs instance. So for example with raw preload=True epochs preload=False your memory consumption will be whatever it takes to store your raw data; for raw preload=True epochs preload=True it will be 2x (assuming your epochs overlap perfectly with all time samples of raw for example).

cbrnr, 2021-02-17:
> My experience with making changes to `Epochs `is that anything involving data loading is never easy and has unintended consequences. If you want to try to work through it I guess we could do it, but I think there are probably many targets more worthy of our attention in our codebase.

OK!

Are you saying that with `Epochs(preload=False)` you need to keep the `raw` object around?

larsoner, 2021-02-17:
> Are you saying that with Epochs(preload=False) you need to keep the raw object around?

It gets stored in `Epochs._raw` or so, so yes it stays around automatically at least until you `epochs.load_data()`

cbrnr, 2021-02-17:
So why can't `EpochsArray` have a `preload` parameter then? Because we pass it data that is already equal to the epoched data, meaning that implicitly `preload=True`?

larsoner, 2021-02-17:
Yes

cbrnr, 2021-02-18:
OK, final thought: would it make sense to make `BaseEpochs` an ABC?

larsoner, 2021-02-18:
What do we gain from it?

cbrnr, 2021-02-18:
Other people won't bother us with these questions again 😄. It's just a matter of style, you don't have to do it, but this is a classic example of an abstract base class, so we could as well declare it as such.

cbrnr, 2021-02-18:
Plus a more tangible benefit is that people can't instantiate `EpochsBase`, they will get an error. Now this is not enforced, it's only in the docstring.

hoechenberger, 2021-02-18:
+1 to make it an ABC if that's a painless process ;)

drammock, 2021-02-18:
As long as you can still type check against it using `isinstance` then I don't care one way or the other.

cbrnr, 2021-02-19:
Type checking with `isinstance` is exactly the point of ABCs, so yes, it works. It should be as simple as inheriting from `collections.ABC` and slapping `@abstractmethod` decorators on methods, but we'll have to see if it is really that painless. I can try and run the test suite to see if this works.

hoechenberger, 2021-02-19:
Sounds good, I'm excited to see how this works!

larsoner, 2021-02-19:
Closing for #8888

closed 2021-02-19 by larsoner