mne-tools/mne-python #8872: Verbose not working with the Epochs.drop_bad method

HanBnrd, 2021-02-15:
#### Describe the bug
The `verbose` argument in not working in the [`drop_bad()`](https://mne.tools/dev/generated/mne.Epochs.html?highlight=drop_bad#mne.Epochs.drop_bad) method.

#### Steps to reproduce
```python
import os
import mne

sample_data_folder = mne.datasets.sample.data_path(path='/db/psxjb6/mne_data')
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=60)
events = mne.find_events(raw, stim_channel='STI 014')
epochs = mne.Epochs(raw, events, tmin=-0.3, tmax=0.7)

epochs.verbose = False

epochs.drop_bad(verbose=True)
```

#### Expected results
The `verbose` argument set to `True` overwrites the `Epochs.verbose` attribute and prints the number of bad epochs dropped.
```
86 events found
Event IDs: [ 1  2  3  4  5 32]
Not setting metadata
Not setting metadata
86 matching events found
Applying baseline correction (mode: mean)
Created an SSP operator (subspace dimension = 3)
3 projection items activated
Loading data for 86 events and 601 original time points ...
1 bad epochs dropped
```

#### Actual results
The number of bad epochs dropped is not printed.
```
86 events found
Event IDs: [ 1  2  3  4  5 32]
Not setting metadata
Not setting metadata
86 matching events found
Applying baseline correction (mode: mean)
Created an SSP operator (subspace dimension = 3)
3 projection items activated
```

#### Additional information
It seems that in the source code of [`drop_bad()`](https://github.com/mne-tools/mne-python/blob/main/mne/epochs.py#L1184-L1224) the argument `verbose` which defaults to `None` is not used.

drammock, 2021-02-15:
I confirm.  @HanBnrd are you up for making a PR to fix this?

HanBnrd, 2021-02-15:
Yes, I'll have a look at what needs to be modified to fix this.

HanBnrd, 2021-02-16:
It seems that this issue is also affecting other methods, such as [`Raw.load_data()`](https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.load_data). The `verbose` argument is not taken into account with the following code:
```python
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=60)

raw.load_data(verbose=True)
```
While this code prints info:
```python
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=60)

raw.verbose = True

raw.load_data()
```
```
Reading 0 ... 36037  =      0.000 ...    60.000 secs...
```
---
Also this following code doesn't print info:
```python
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=60)

mne.set_log_level(verbose=True)

raw.load_data()
```
Is the [`mne.set_log_level()`](https://mne.tools/dev/generated/mne.set_log_level.html#mne.set_log_level) function supposed to work in this case?

HanBnrd, 2021-02-16:
This issue seems to be related to the [`@verbose`](https://github.com/mne-tools/mne-python/blob/main/mne/utils/_logging.py#L58-L135) decorator not working in some cases (maybe working on functions or methods calling functions but not on proper methods?)...

An easy quick fix would be to explicitly use the `verbose` arguments in the methods instead of relying on `@verbose` but it is probably not a good long term solution to solve the issue on all the methods affected.

larsoner, 2021-02-16:
> Also this following code doesn't print info:

That one at least shouldn't based on how the code is designed. The precedence/order is meant to be:

- `verbose` arg is used if not None
- `verbose` attribute of a class is used if not None
- global verbose level is used

And in that last example, the `raw` gets initialized with `verbose=False`, so `raw.load_data()` should always be in non-verbose mode regardless of `set_log_level`.

larsoner, 2021-02-16:
(I can take a look at fixing the `verbose` decorator if it would help, since it does seem like the first/top-comment example is problematic)

HanBnrd, 2021-02-16:
> 
> 
> (I can take a look at fixing the `verbose` decorator if it would help, since it does seem like the first/top-comment example is problematic)

Thanks @larsoner, that would be nice if you have time as I'm not very familiar with it and it appears to be used as a decorator for a lot of functions/methods in MNE. Please let me know if I can help.

PR #8884 (2021-02-17): MRG, BUG: Fix verbose
Closes #8872

@HanBnrd can you try it with your code?

Basically the fix is that for any methods that have `verbose`, we need to propagate `verbose` to any other method calls. We don't need to do this with functions, but we do with methods since when they are called with `verbose is None` they resort to using `self.verbose`.
closed 2021-02-17 by larsoner