mne-tools/mne-python #8877: How to reject bad segments when working with long epochs

KathrineSN, 2021-02-16:
I am doing a social cognition experiment with eeg for my thesis, and in my experiment I am working with long trials of 25 seconds. 
This means that if I use any automatic rejection function to reject bad segments, it will remove a large amount of my data as the entire epoch is being dropped when using both mne.annotations or autoreject. Any suggestions on how to remove bad segments in my 25 seconds long epochs without removing entire epochs?

All help is much appreciated,

Kathrine

larsoner, 2021-02-16:
Could you make 1 sec epochs instead, 25 in a row of the given type? This is how I've dealt with this sort of problem in the past.

KathrineSN, 2021-02-17:
Yes, that is a good idea. The only problem would then be to merge the 25 1 sec epochs back into one epoch before I have to analyze the data...

larsoner, 2021-02-17:
This seems more like a usage question at this point so could you bring this up on Discourse? Then others can weigh in with ideas and experiences. I'll close this since I don't think there is anything to change at the code end (yet).

closed 2021-02-17 by larsoner