mne-tools/mne-python #8878: Additional functions to take the "average" of EpochsTFR?

adam2392, 2021-02-16:
#### Describe the new feature or enhancement
In https://mne.tools/stable/generated/mne.time_frequency.EpochsTFR.html#mne.time_frequency.EpochsTFR.average one can currently compute the average of the TFR over all Epochs. This simply calls the numpy.mean function. 

Would it be possible to add additional combination possibilities, such as `np.median`, `np.percentile`, such that `average(func)` is now possible, with the list of funcs being possible are numpy functions that operate along the epochs axis.

#### Describe possible alternatives
One can always compute this themselves, I suppose, but then they lose access to `AverageTFR` api functionality.

agramfort, 2021-02-16:
Sure

See how it’s exposed in epochs.average

adam2392, 2021-02-16:
I can then adjust these lines accordingly? 

https://github.com/mne-tools/mne-python/blob/4df9ed838f1d56471d071308fb62b9ec749cb10d/mne/time_frequency/tfr.py#L2092-L2104

- Should I keep the `average()` function name as is and just allow optional kwarg `func`? 

- Should I also specify how the "average" was taken in the `AverageTFR.comments`? So if I use np.median instead of np.mean, i'll pass in `AverageTFR(..., comments=self.comment + ' - took average via np.median')`?

agramfort, 2021-02-16:
Copy how it’s done in epochs including api

PR #8879 (2021-03-02): [MRG] Add optional different ways of averaging EpochsTFR
#### Reference issue
Closes: #8878 

#### What does this implement/fix?
Essentially same workflow as `Epochs.average()`.
closed 2021-03-02 by agramfort