mne-tools/mne-python #8880: Implementation of stft() function as stft_array and tfr_stft

adam2392, 2021-02-16:
#### Describe the new feature or enhancement
Currently `stft` operates on arrays, but is not the same API as wavelet, stockwell and multitaper. In addition, it does not have a `tfr_stft` analog , which operates on mne-python data structures.

The proposed PR would refactor `stft` to be named `stft_array` and have similar API to the `tfr_array` functions, and also implement an analogous `tfr_stft` function.

See discourse discussion: https://mne.discourse.group/t/is-there-a-way-to-obtain-the-time-points-from-time-frequency-stft/2682

The `stft` is a nice function since it is super fast, and can operate on `Raw` object data within seconds. 

#### Describe your proposed implementation
Tbd

#### Additional comments
related: https://github.com/mne-tools/mne-python/pull/8781. If this is accepted, then one can use `tfr_stft` to compute spectrograms of the data for certain frequency bands (e.g. gamma power shown as channels X time over a whole Raw dataset).

larsoner, 2021-02-17:
> The proposed PR would refactor stft to be named stft_array and have similar API to the tfr_array functions, and also implement an analogous tfr_stft function.

I would propose instead that we deprecate `stft` (have it call `_stft` that has all the current `stft` code in it) and make a new `stft_array` that has the right API (but calls `_stft` under the hood). I don't think we ever really meant for `stft` to be public, right @agramfort ?

adam2392, 2021-02-17:
And should I still include `tfr_stft`?

larsoner, 2021-02-17:
Yeah that would make sense to fit with the existing API

agramfort, 2021-02-19:
ok to move replace stft to _stft and create new public functions tfr_stft
and tfr_array_stft that just calls _stft?
that's what you had in mind?

>