mne-tools/mne-python #8881: CI: make CIs use environment/requirements files?

drammock, 2021-02-16:
I seem to keep hitting this problem that the CIs don't actually use our environment/requirements files (instead they use bespoke things like `.github/workflows/*.yml` or `tools/github_actions_dependencies.sh`).  Why is this?  Why does adding a dependency have to be done in like six places?  It's bad enough to have to remember to edit `environment`, `server_environment`, and `requirements`.  Is there any possibility to make this more dev-user-friendly?

larsoner, 2021-02-16:
It's because:

1. We have optional dependencies. Our environment.yml / requirements.txt are meant to be "feature-complete", i.e., include all optional ones required to have all functionality (except CUDA because it's a weird one).
2. We have to test old versions of dependencies. We would never want to force our users to install these, but some have old versions of libraries, so we need to test that we actually comply with our minimum version spec in the readme.

One way out to at least make things more consistent would be to move the logic mostly out of the CI files and into new files like `environment_old.yml` / `requirements_old.txt` and/or `environment_minimal.yml` / `requirements_minimal.txt`. Then the idea is that if you have to add a dependency, you "just" need to add it to all `environment*` and `requirements*` files, depending on the extent to which they are optional etc. But I'm not sure how much easier this will really make things.

Our pattern/mantra has pretty much been "try hard not to add dependencies" up to this point, so it hasn't been too much of a problem / bottleneck, even if it's not the most straightforward thing.

drammock, 2021-02-17:
WDYT about putting it all in one YAML or py file and having some sort of pre-commit hook that generates the various other dependency files?

agramfort, 2021-02-17:
> WDYT about putting it all in one YAML or py file and having some sort of pre-commit hook that generates the various other dependency files?
> 

We very rarely add dependencies so this seems to me over engineering

drammock, 2021-03-03:
I still think this is a good idea, but too low a priority to realistically get done any time soon.