mne-tools/mne-python #8883: Tests fail on maint/0.22

cbrnr, 2021-02-17:
After backporting #8874, various tests now fail. I suspect that because we now use `testing` 0.116 (previously it was 0.112), several test files other than the one I added in 0.116 changed, and tests have been adapted in `main` only.

Is there an easy way to find out which additional changes need to be backported? Otherwise, I'd revert my last 3 commits...

@larsoner @hoechenberger any suggestions?

hoechenberger, 2021-02-17:
Oh that's super unfortunate! I would say, just revert it.... and we should push out 0.23 sooner than later 😎

larsoner, 2021-02-17:
One option would be to remove the test that needs the 0.116 data, revert the testing data uptick, but keep the actual fix in the EEGLAB code. We don't usually do this but it seems okay in this case as a workaround to keep the fix (especially knowing the fix was not overly complicated)

cbrnr, 2021-02-17:
Good idea, I'll try that @larsoner.

cbrnr, 2021-02-17:
Let's see if this works: https://github.com/mne-tools/mne-python/commit/535c11093ae5dcecd2e33032e428fe4c77dcfb3c

larsoner, 2021-02-17:
Looks like you reverted the number but not the hash

cbrnr, 2021-02-17:
🤦

cbrnr, 2021-02-17:
https://github.com/mne-tools/mne-python/commit/b77d479ed6c0020d83e2fe36f3eb39ae8da0ebdd

cbrnr, 2021-02-17:
@larsoner [Windows 3.8 pip](https://dev.azure.com/mne-tools/mne-python/_build/results?buildId=11702&view=logs&j=a017e066-62ca-5289-ad0b-8f57c84a089f&t=de70cabd-1dad-599c-0751-4f1f50c17e0f) fails big time with a bunch of `FutureWarnings` in `numpy`...

larsoner, 2021-02-17:
Yes after a while after release the `pip pre` builds inevitably have errors, we can ignore them in this case at least

cbrnr, 2021-02-17:
Uhm - why? After an MNE release you mean? I don't get it.

In any case, I guess this means that `maint/0.22` is OK again?

larsoner, 2021-02-17:
> Uhm - why? After an MNE release you mean? I don't get it.

The `pip pre` builds test against *prelease* versions of libraries like NumPy dev and SciPy dev etc. Usually:

1. Their changes require updates to our code once or twice a month to comply properly, which means inevitably our release falls behind and starts failing. It is fixable (we fix them in `main` after all) but...
2. Those libraries have long enough release cycles, we have fast enough releasy cycles, and people have slow enough upgrade patterns that it's often not worth the effort to avoid what is usually a harmless DeprecationWarning someone might hit.

closed 2021-02-17 by cbrnr