mne-tools/mne-python #8885: Different file size limits for fif files in Neuromag and mne

eort, 2021-02-17:
Was it intentional to set the file size limit to precisely 2gb? It seems that the Neuromag Vectorview uses a limit of 2gb minus 100mb (although this is configurable, see page [54](http://megmoscow.ru/resources/access/equipment/DataAcquisition.pdf) of the user manual). No clue whether this is common for other Neuromag's scanners.

I'm wondering because some of my recordings happen to be just below 2gb, so that the MEG scanner split the files, but when running the data through `mne-bids`, I ended up with a single file. Not that this is a big problem, but when I tried to figure out based on a `raw` instance whether a file would be split once saved, I relied on the `raw.filenames` attribute. In this particular edge case, that approach doesn't work.

larsoner, 2021-02-17:
We actually use 2 GB minus some slop, too. It's just our slop is a lot less than 100 MB -- we do some calculations to guess what `info` and other items require and then add some additional buffer like 10 MB or so to deal with the end-of-file directory writing (I think that's what it is anyway?). In general I wouldn't expect / rely on our implementation and theirs ever matching, so it's best to write code that's robust to split sizes changing depending on who is doing the writing.

> That being said, is there a way to predict how large a raw object will be after saving on disk?

There is some private code somewhere in the depths of `mne/io/base.py` that would do this sort of computation. If it helps to make this public, I guess in principle we could. But how would this help your use case?

eort, 2021-02-17:
Ah, I see. Thanks for clarifying. 

>  But how would this help your use case?

It's related to [this discussion](https://github.com/mne-tools/mne-bids/issues) at the mne-bids repo. In short, split files are currently not listed separately in the `scans.tsv`, which they should. In order to list each split file separately, you need to know whether a file will be split, obviously. But as far as I see, it is only determined during the actual process of saving whether a file is split or not. That's why I went the detour over the number of inputfiles in `raw.filenames`.