mne-tools/mne-python #8890: BUG: Channel-Adjacency for Neuromag122

marsipu, 2021-02-20:
#### Describe the bug
When I try to get the channel-adjacency-matrix for Gradiometer from **Neuromag122**-Data with `find_ch_adjacency`
, a matrix with dimensions 306x306 is given (when read from ./mne/channels/data/neighbors/neuromag122_neighb.mat).

And when I try to compute the adjacency with `mne.channels.channels._compute_ch_adjacency`, I get the following result:

```
File "C:\Users\marti\anaconda3\envs\mne\lib\site-packages\mne\channels\channels.py", line 1426, in _compute_ch_adjacency
xy = _find_topomap_coords(info, picks, sphere=HEAD_SIZE_DEFAULT)
File "C:\Users\marti\anaconda3\envs\mne\lib\site-packages\mne\channels\layout.py", line 624, in _find_topomap_coords
pos = _auto_topomap_coords(
File "C:\Users\marti\anaconda3\envs\mne\lib\site-packages\mne\channels\layout.py", line 723, in _auto_topomap_coords
raise ValueError('The following electrodes have overlapping positions,'
ValueError: The following electrodes have overlapping positions, which causes problems during visualization:
MEG 001, MEG 002, MEG 003, [...]
```

#### Steps to reproduce
```
import mne
data_path = "./neuromag122_test_file-raw.fif"
info = mne.io.read_info(data_path)
# Trying the load the channel-adjacency
ch_adjacency = mne.channels.find_ch_adjacency(info, 'grad')
print(ch_adjacency[0].shape)
# Trying to bypass mne.channels.read_ch_adjacency()
mne.channels.channels._compute_ch_adjacency(info, 'grad')
```

#### Expected results
A 122x122 Matrix

#### Actual results
A 306x306 Matrix and a Value Error

#### Additional information
I uploaded an [anonymized raw-file](https://github.com/mne-tools/mne-python/files/6016694/neuromag122_test_file-raw.zip) from our Neuromag122 for reproduction of the issue.

marsipu, 2021-02-20:
I dug into the problem and other functions like plots are working so probably the problem lies with how the paired gradiometer-channels in Neuromag122 are handled by _compute_ch_adjacency. For the info of my test-file, line 1412 in mne/channels/channels doesn\'t get True. The Coil-Type of Neuromag122 is FIFFV_COIL_NM_122

marsipu, 2021-02-20:
I will open a draft PR with a possible fix, but I still don't know why the saved matrix from fieldtrip in `/channels/data/neighbours` is 306x306

larsoner, 2021-02-22:
> When I try to get the channel-adjacency-matrix for Gradiometer from Neuromag122-Data with find_ch_adjacency, a matrix with dimensions 306x306 is given (when read from ./mne/channels/data/neighbors/neuromag122_neighb.mat).

I think we got this file from FieldTrip -- looks like there was a similar error about 8 years ago. I've opened and upstream issue: https://github.com/fieldtrip/fieldtrip/issues/1676

larsoner, 2021-02-22:
Sound like the correct fix is to kill the `.mat` file, can you remove it from the repo using `git rm` and remove the commented-out lines in this PR?

larsoner, 2021-02-22:
Sorry meant for that comment to land in #8891

PR #8891 (2021-03-02): Fix: Wrong channel-adjacency-matrix for Neuromag122
#### Reference issue
Fixes #8890.

#### What does this implement/fix?
This disables reading the channel-adjacency for Neuromag122-Data from a precomputed file in .../channels/data/neighbors and fixes the computation of the matrix.
closed 2021-03-02 by larsoner