mne-tools/mne-python #8897: ValueError reading Epochs .fif file saved using the inbuilt Epochs.save() method with mne.read_epochs()

rtybanana, 2021-02-22:
### Describe the bug
I opened raw GDF data and performed a few preprocessing methods on it: ICA, bandpass filtering and resampling. I then Epoched the Raw data and saved it back into a .fif file so that instead of doing the preprocessing again I could just open the file directly into an Epochs object. However, when attempting to load it in a `ValueError` is raised:

``` Python
ValueError: Could not find event data
```

### Steps to reproduce
Save Epoch object to file with .save() and attempt to open it again with mne.read_epochs()

### Expected results
Epochs object successfully created with all of the data inside

### Actual results
`ValueError` raised because event data could not be found.

larsoner, 2021-02-23:
> Save Epoch object to file with .save() and attempt to open it again with mne.read_epochs()

I can't reproduce with `sample` data:
```
>>> raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif')
>>> epochs = mne.make_fixed_length_epochs(raw)
>>> epochs.save('test-epo.fif')
>>> epochs = mne.read_epochs('test-epo.fif')
>>> epochs.save('test-epo.fif', overwrite=True)
>>> epochs = mne.read_epochs('test-epo.fif')
```
So we'll need to work with your data. Can you upload the raw GDF file and a script to reproduce the error?

rtybanana, 2021-02-23:
While writing out the example script I discovered my mistake which was that I was saving the Raw object rather than the Epochs and trying to load it with read_epochs.

Perhaps a more informative error message further up would be useful? Though I doubt this happens enough to warrant it being a priority change.

Thanks for your help.

closed 2021-02-23 by rtybanana