mne-tools/mne-python #8930: use bibtex in multi_comp.py

drammock, 2021-02-23:
convert references in `mne/stats/multi_comp.py` to use footcite / footbibliography

FinLouarn, 2021-03-15:
I'm taking this one

drammock, 2021-03-15:
You can add this to the bibtex file.  Try to insert it roughly in the right spot for alphabetical order by first author surname (not strictly required, but makes it easier to scan through)

```
@article{GenoveseEtAl2002,
  title = {Thresholding of {{Statistical Maps}} in {{Functional Neuroimaging Using}} the {{False Discovery Rate}}},
  author = {Genovese, Christopher R. and Lazar, Nicole A. and Nichols, Thomas},
  year = {2002},
  month = apr,
  volume = {15},
  pages = {870--878},
  issn = {10538119},
  doi = {10.1006/nimg.2001.1037},
  journal = {NeuroImage},
  language = {en},
  number = {4}
}
```

PR #9055 (2021-03-19): bibtext and reference.bib updated for 'Correction for multiple comparison using FDR' in multi_comp.py

#### Reference issue
Fixes #8930 .

#### What does this implement/fix?
updated bibtext and references.bib for 'Correction for multiple comparison using FDR' in multi_comp.py

#### Additional information
The html doc have been built properly locally
closed 2021-03-19 by sappelhoff