mne-tools/mne-python #8950: ENH: Repair channels on a per-epoch basis

larsoner, 2021-02-24:
Now that we have #8896 we can redo #1869 such that we can repair channels on a per-epoch basis.

The challenge I think will be making it efficient, but I think this will actually not be too bad if we refactor the interpolation code:

1. Do the slowest part once: self- and cross-dots between all channels
2. Do the faster part each time we need to: pseudoinversion based on all "from" good channels to all "to" bad channels for the given epoch
3. Use a simple, small LRU cache to store the from-to paired pinv results (if you use more than 100 different "bad" combinations, which should be very rare, you start to pay a price for having to redo some of the pinv calls, but even this probably won't be too bad).

In principle someday we could do this on raw data, but we have to figure out how to prevent discontinuities between the original and reconstructed data segments. This should be doable with some suitable smooth windowing but it's not as easy as the epochs case.

jasmainak, 2021-02-24:
There is code in autoreject that does the caching. It would be great if we can copy it and autoreject uses the public API of MNE

agramfort, 2021-02-24:
Having this in mne will simplify a lot the auto reject code !

larsoner, 2021-10-14:
I guess the place to start would be with a PR to take care of #8951 first