mne-tools/mne-python #8951: ENH: API for handling channel-specific annotations in epoching

larsoner, 2021-02-24:
One nice pipeline is:

1. Mark channel-specific annotations (#8947 / #8949)
2. Epoch the data without throwing out epochs based on channel-specific annotations (this ENH request)
3. Preserve the annotations from Raw->Epochs (#8376)
4. Repair channel-specific bads (#8950)

To be able to do 4/#8950 we really want some API for creating epochs that probably does throw out epochs where a given time slice is marked as bad, but (optionally) does not throw out epochs where there are channel-specific BAD_ markings.

I actually don't think this one is too difficult, but it requires at least #8376 (and probably a preprocessing one like #8947 / #8949) to move forward.

cbrnr, 2021-02-24:
Would repairing also work when there are all-channel bad annotations? Currently, I sometimes have rather short artifact segments in relatively long epochs, and it is wasteful to discard the whole epoch just because of said artifacts. Is this possible right now, and if not, could we add support? For example, subsequent methods could ignore such segments and only work with the "good" data in an epoch (e.g. for averaging this would be relatively straightforward, and it should work for filtering as well, but there might be other more complicated methods where this doesn't work).

larsoner, 2021-02-24:
> Would repairing also work when there are all-channel bad annotations?

No, that would require a different algorithm. This issue and #8950 are really about channel-specific repair which just uses spatial interpolation. What you're talking about would require some sort of temporal interpolation for all channels. Feel free to open a separate issue to discuss this, as it's not clear what algorithm to use (we don't have any in MNE AFAIK).