mne-tools/mne-python #8952: Fix position of TOCs (left and right columns)

hoechenberger, 2021-02-24:
The tables of content displayed in the left and right columns of the development documentation should be remain in view as I scroll down the page; currently, they are affixed to the top of the page (not top of the window), meaning they'll disappear as you start scrolling down, which makes it unnecessarily difficult to keep track of which part of the document you're currently looking at.

drammock, 2021-02-24:
+1 for this, but I think this is a change that might be valuable upstream.  I'd suggest making a PR into pydata-sphinx-theme with "sticky" sidebars as a theme configuration option set in `conf.py`

larsoner, 2023-10-01:
I think we have this now but feel free to reopen if I'm misunderstanding

closed 2023-10-01 by larsoner