mne-tools/mne-python #8958: Justify content of center column

hoechenberger, 2021-02-25:
When reading through longer tutorials, sometimes the left-aligned text in the center column makes for a rather unpleasant visual appearance:
[image]

By adding Bootstrap's `text-justify` class to the respective container; however it's still lacking proper hyphenation:

[image]

Adding the following CSS enables hyphenation across different browsers:

```css
  word-break: break-word;  /* Chrome */
  -webkit-hyphens: auto;
  -moz-hyphens: auto;
  -ms-hyphens: auto;
  -o-hyphens: auto;
  hyphens: auto;
```

[image]

One would need to ensure that this is not applied to `DataFrame` renderings and headlines.

Again, I'm not sure whether such a change should be made here or upstream?

drammock, 2021-02-25:
-1 for this change. Although justification looks nice to some people, studies show that it actually slows down reading.

hoechenberger, 2021-02-25:
@drammock is right: It appears that for LTR languages, left alignment is to be preferred, as it provided better accessibility. 

https://webaim.org/techniques/textlayout/

closed 2021-02-25 by hoechenberger