mne-tools/mne-python #8974: Add HED to glossary

hoechenberger, 2021-03-01:
In MNE, we can describe events using hierarchical descriptors, such as `stim/a`, `stim/b` etc., named hierarchical event decriptiors (HED). They should be mentioned in the glossary and potentially a reference to the original publication should be added to an Epochs/events tutorial

Publication:
http://sccn.ucsd.edu/~nima/downloads/HED_globalsip_paper.pdf

sappelhoff, 2021-03-04:
:+1: 

There is also a more recent paper: https://osf.io/5fg73/

the initial paper talks about "first generation HED", but the standard is planning to release the third generation in 2021.

hoechenberger, 2021-03-04:
Yes but I don't think we currently support anything beyond HED "1.0", do we?

sappelhoff, 2021-03-04:
> Yes but I don't think we currently support anything beyond HED "1.0", do we?

I am not sure that MNE supports HED 1.0 fully :thinking: it looks to me more like some (more or less coincidental) overlap between features, but I may be wrong.

anaradanovic, 2021-03-15:
I can work on this one! :)

hoechenberger, 2021-03-15:
@anaradanovic This sounds great!

anaradanovic, 2021-03-15:
Would you like me to include examples in the description?

hoechenberger, 2021-03-15:
@anaradanovic Yes why not!

PR #9097 (2021-03-20): DOC: Add HEDs to glossary and reference in tutorial and epochs.py  #8974.
#### Reference issue
Fixes #8974.

#### What does this implement/fix?
I added the citation for hierarchical event descriptors in the glossary, added this citation to the tutorial (plot_40_autogenerate_metadata.py ) and to epochs.py. 
Also fixed a couple typos in the description in plot_40_autogenerate_metadata.py.
closed 2021-03-20 by drammock