mne-tools/mne-python #8976: Add support for file-like objects to mne.io.read_raw_edf()

CJ-Wave, 2021-03-01:
#### Describe the new feature or enhancement
I would like to have support for file-like objects for the mne.io.read_raw_edf() function (specifically a bytesIO object)

#### Describe your proposed implementation
new capability to an existing method

#### Describe possible alternatives
Possible alternative is allowing for a kwarg to detect if it's a file-like object as opposed to a filepath, but shouldn't be necessary

#### Additional comments
It looks like fiff files have that functionality in mne.io.read_raw(), it would be fantastic if edf had that functionality as well

larsoner, 2021-03-01:
Agreed this would be useful and I don't see any reason offhand why it would be a problem. Maybe some code in `mne/io/fiff/raw.py` could be refactored for this. Do you want to give it a try @CJ-Wave ?

CJ-Wave, 2021-03-01:
Sure, I can give it a try.

pb2640, 2023-02-04:
> Sure, I can give it a try.
Hi @CJ-Wave, I was wondering if you were able to implement this feature?

szz-dvl, 2025-03-11:
Hi, I'd love this functionality too, is there any plan to implement it soon? may I help somehow?

Thanks

larsoner, 2025-03-11:
@szz-dvl I don't tihnk anyone else is working on it, help is welcome!

szz-dvl, 2025-03-12:
@larsoner I'm trying to test my changes for GDF files, I downloaded test data from [here](https://github.com/mne-tools/mne-testing-data/tree/master), most of the tests are failing with `io.UnsupportedOperation: fileno`, the error comes from the calls to `np.fromfile` in the function `_read_gdf_header`. By changing `np.fromfile` to `np.frombuffer` and adapting the parameters I can get rid of the error, but, I'm wondering if somebody can check this in another environment before performing this changes, I'm not sure if it is a numpy version issue, I'm using numpy 2.2.3.

Thanks!

larsoner, 2025-03-12:
@szz-dvl I would expect a change like that to be needed, yeah. The easiest thing to do is probably to open a pull request with your changes. The CIs will then run tests on all OSes and we'll see if it works and what (if anything) breaks!

szz-dvl, 2025-03-12:
@larsoner Please forget about np.frombuffer, it was on my side.

Forgive any inconvenience.