mne-tools/mne-python #8979: Add add_reference_channels() method to Raw, Epochs, and Evoked

hoechenberger, 2021-03-02:
Currently there's `mne.add_reference_channels()`, which takes an instance of `Raw`, `Epochs`, or `Evoked`, and adds reference channels.

It would be great to have this functionality exposed as a **method** too, i.e., `Raw.add_reference_channels()`, `Epochs.add_reference_channels()`, and `Evoked.add_reference_channels()`, just as is the case e.g. with `set_eeg_reference()`

The respective `Raw`/`Epochs`/`Evoked` instance should be modified in-place, and the modified instance should be returned to allow for method chaining.

This would need to be added to `UpdateChannelsMixin` in `mne/channels/channels.py`

Overlapping parts of the method and  `mne.add_reference_channels()` docstrings should be added to the `docdict` to avoid duplication.

agramfort, 2021-03-02:
no objection

>

Falach, 2021-03-16:
I'll take this one :)

PR #9135 (2021-03-21): adding add_reference_channels as a method of Raw/Epochs/Evoked
#### Reference issue
Fixes #8979

#### What does this implement/fix?
add_reference_channels is now also a method of Raw/Epochs/Evoked

#### Additional information
I also implement a little test that checks mainly that the object has changed (the return value is not a copy), there are more tests already at the original function location (mne/io/reference).
Let me know if the docdict is used as you expected.
closed 2021-03-21 by agramfort