mne-tools/mne-python #8982: RFC: Set ICA's max_iter to a higher default

hoechenberger, 2021-03-03:
Currently, `ICA.max_iter` is set to 200 by default. This is one of the few MNE defaults I **always** change when working with data from our lab, and I've seen quite a few support requests indicating that ICA didn't converge within those 200 iterations too, the most recent one here: https://mne.discourse.group/t/cannot-find-ecg-artifacts-during-ica/2778/11?u=richard

I'd like to suggest to increase `max_iter` to 1000 to ensure that non-convergence will really be an exception. I don't think there's many valid use cases for working with non-converged ICA decompositions anyway, or are there? Every time I've encountered the warning about non-converged ICA, the decomposition looked … flaky.

WDYT?

cc @cbrnr

agramfort, 2021-03-03:
how is it done in other software packages?

>

hoechenberger, 2021-03-03:
> how is it done in other software packages?

EEGLAB uses

[500 for picard](https://github.com/sccn/eeglab/blob/96f5b56adec802ea3fc296085f17d3f21c204942/functions/popfunc/pop_runica.m#L167)
[1000 for pearson ICA](https://github.com/sccn/eeglab/blob/96f5b56adec802ea3fc296085f17d3f21c204942/functions/popfunc/pop_runica.m#L487)

For Infomax, AMICA and FastICA I couldn't find any defaults set in the EEGLAB sources – they're probably relying on the respective defaults of the binaries they're calling..?

agramfort, 2021-03-03:
let's try to stick to the same defautls

>

hoechenberger, 2021-03-03:
Ok it seems to be
[1000 for FastICA](https://github.com/davidkun/FastICA/blob/df9dbda6a50161285afd6933b04241f974c6e377/fastica.m#L94-L95)
[2000 for AMICA](https://github.com/japalmer29/amica/blob/008abb83a98ed34c0c57785804ee34ec52a59c4b/pop_runamica.m#L10)

Infomax I still couldn't find.

hoechenberger, 2021-03-03:
[512 for (extended) Infomax](https://github.com/sccn/eeglab/blob/96f5b56adec802ea3fc296085f17d3f21c204942/functions/sigprocfunc/binica.m#L29)

hoechenberger, 2021-03-03:
--> My proosal:

- Change the signature of `ICA(..., max_iter=200)` to `... max_iter=None`
- fill in the default based on the used algorithm:
    - Infomax & picard -> 500
    - fastica -> 1000

Would also be suitable as a task for the sprint if we agree this is a change we'd like to have.

agramfort, 2021-03-03:
ok for a sprint issue

>

kimcoco, 2021-03-15:
I will take this one, and then move on to other ICA issues w/ Tutorial, Example

hoechenberger, 2021-03-15:
Perfect!

PR #9099 (2021-03-18): ICA max_iter new defaults
#### Reference issue
Example: Fixes #8982.

#### What does this implement/fix?
I changed the default max_iter values for different ICA methods - as proposed in the issue. Also added a test function.

#### Additional information
closed 2021-03-18 by hoechenberger