mne-tools/mne-python #8991: set_eeg_reference to multiple channel types

AlexLepauvre, 2021-03-04:
# Issue
I am analyzing data sets consisting of both ECoG and SEEG channels. One of our preprocessing step is to apply CAR the ecog and seeg channels. In order to ensure comparability between the different electrodes (regardless of whether they are seeg or ecog), we need to apply CAR to all ecog and seeg channels at once, the average to apply also being across seeg and ecog channels. While the set_eeg_reference enables to state which channels should be taken to compute the average, it is only possible to set one channel type as which channels to apply it to. 

I have been doing it sequentially at first:
<raw.set_eeg_reference(projection=False, ref_channels=ref_channels, ch_type='seeg')>
then:
<raw_sig.set_eeg_reference(projection=False, ref_channels=ref_channels, ch_type='ecog')>
(ref_channels being all seeg and ecog channels here)

This is obviously wrong, as when the second statement is reached, CAR has already been applied to the seeg channels, making the second round of CAR wrong.

# Proposed implementation
Would it be possible to pass a list of channel type instead of a single string. 

Alternatively, would it be possible to pass a list of channels to apply the CAR to?

# Additional comments
Apologies if this a misunderstanding of how things work, I am still rather new to mne. I would offer to handle it myself and do a PR, I am however very uncertain that my programming skill fullfil the mne standards.

agramfort, 2021-03-05:
indeed we had not considered this use case.

I see 2 solutions for now:
- temporarily change ecog channels to seeg to be able to set the ref (if it
works)
- see if https://mne.tools/stable/generated/mne.set_bipolar_reference.html
which accepts a list
of channel names can be used.

HTH

>

larsoner, 2021-03-05:
We should allow `ch_type` to be a `list of str` to support this use case. This seems like a fairly small change, with little risk of users shooting themselves in the foot (they have to explicitly pass such a list, so presumably know what they want).

> I would offer to handle it myself and do a PR, I am however very uncertain that my programming skill fullfil the mne standards.

We all start somewhere, and we're happy to help you along. And if you get stuck or run out of time to work on it, maintainers can push commits to your branch. So a PR would be welcome if you're up for it!

AlexLepauvre, 2021-03-10:
Sorry for the late reply.
I will give it a go right now. I will try to issue the PR by the end of the week.

mscheltienne, 2021-08-11:
To be closed.