mne-tools/mne-python #8994: Append functionality for AverageTFR

adam2392, 2021-03-04:
#### Describe the new feature or enhancement
Similar to how Raw instances can be appended together, I was wondering if it is possible to include that for AverageTFR. Running instances separately, and it's useful for me to append these together in my workflo.w

#### Describe your proposed implementation
Essentially take Raw.append and copy it over with additional checks on the frequency axis.

agramfort, 2021-03-04:
why do you need this? can you append the raw and compute on this?

adam2392, 2021-03-04:
My workflow rn consists of different recording sessions. I’m running a tfr spectral analysis and saving output as mne h5 files. 

But when I do some post analysis I want to join them into one contiguous chunk to explore how the freq content changes.

agramfort, 2021-03-05:
can you cook up such a function in our own code base while you develop the
use case
and the results and we consider it adding this to MNE in the future?

>