mne-tools/mne-python #9002: convert examples/preprocessing/plot_ica_comparison to full tutorial

drammock, 2021-03-06:
This example has the potential to be a more informative discussion about the pros/cons of different ICA algorithms.  Also right now it doesn't really fit the goal of being a "How-to" for a normal analysis task (you wouldn't usually compare several ICA algorithms every time you preprocess a new dataset).  Things it should discuss:

- speed
- performance on clean data vs different kinds of noisy data
- deterministic / random?

would be good to also show the "same" component from different algorithms side-by-side (right now just a separate grid of 25 components for each algorithm)

cmista, 2021-03-18:
I want to take this one.

Ganasekhar-gif, 2025-04-06:
Hi! 

I'm a student currently preparing a GSoC proposal for the MNE-Python project and am very interested in contributing to this issue. Converting the ICA comparison example into a full tutorial aligns well with my project idea, which focuses on enhancing the MNE tutorial experience, particularly in areas like preprocessing and ICA.

I'd love to take up this issue and start working on a PR. Please let me know if there are any specific guidelines or expectations you'd like me to follow while working on this.

Looking forward to your feedback!

drammock, 2025-04-08:
> I'd love to take up this issue and start working on a PR. Please let me know if there are any specific guidelines or expectations you'd like me to follow while working on this.

feel free to take a swing at it! The original description of this issue should provide good starting guidance, and feel free to open a work-in-progress PR in draft mode and ask any clarifying questions there.

Ganasekhar-gif, 2025-04-09:
Thank you! I’ll go through the original issue description in detail and start working on a draft PR soon. I’ll make sure to follow the existing structure and style used in the tutorials, and I’ll reach out if I have any questions or need clarification along the way.

Appreciate the support—looking forward to contributing!

Ganasekhar-gif, 2025-04-16:
Hi drammock! Just a quick follow-up — I’ve opened a draft PR for this issue  and included a few questions there regarding design decisions and clarification. I’d love to get your feedback whenever you have time, especially to make sure I’m on the right track before moving further. Thanks a lot!

drammock, 2025-04-25:
apologies for the delay, I haven't forgotten about this. Will look it over as soon as I'm able.

Ganasekhar-gif, 2025-05-16:
Hi @drammock,
Just following up on the draft PR I opened a little while ago for this issue. I completely understand things can get busy, but I’d really appreciate it if you could take a look whenever you have the time. Your feedback would be really helpful to ensure I’m on the right track before proceeding further.

Thanks again, and no rush — just wanted to check in!