mne-tools/mne-python #9004: split code block in examples/preprocessing/plot_virtual_evoked

drammock, 2021-03-06:
right now, because all plots come from a single code block, they are plotted at the top of the example in a group of 4 (and consequently the plots are really small).  By splitting the 4 plotting calls into different code blocks, they will plot larger / be easier to see & compare, without increasing run time of the example.  Code blocks can be split with a line of 79 `#` marks (adding a bit of explanatory text too is usually a good idea)

eort, 2021-03-16:
I would like to take care of this one

eort, 2021-03-16:
File name (plot_evoked) and title (Remap mag channels) of this example are different. Not sure how important it is to you that they match?
In my opinion, the title seems to be more appropriate than the file name, but changing file names is probably more troublesome than changing a title.

PR #9092 (2021-03-16): split up code cells in example
#### Reference issue
Fixes #9004

#### What does this implement/fix?
- the code was split into four code blocks, so that the figures are printed in separate cells. 
- a few lines of explanation were added
closed 2021-03-16 by agramfort