mne-tools/mne-python #9007: Fix corrupted figures in plot_background_freesurfer_mne

drammock, 2021-03-06:
Several plots in `tutorials/source-modeling/plot_background_freesurfer_mne.py` are messed up (inverted, misaligned, or corrupted).  See if you can fix them. Check with @larsoner or @GuillaumeFavelier before you start; some of them might be due to 3D viz bugs that aren't fixed yet.

larsoner, 2021-03-08:
I think this has to do with a change related to whether the camera's focal point is at 0, 0, 0 (new behavior) or center of the bounding box of all objects in the scene (old behavior). @GuillaumeFavelier I think we should make sure that there is some option like `focalpoint='bounds'` or so (and focalpoint=None should mean "use existing focal point"), and maybe set "bounds" (back) to the default. WDYT?

GuillaumeFavelier, 2021-03-08:
>  this has to do with a change related to whether the camera's focal point is at 0, 0, 0 (new behavior) or center of the bounding box of all objects in the scene (old behavior)

Agreed, the behaviour of the focal point should be clarified and maybe even tested (to avoid this out of focus brain regression) ?

larsoner, 2021-03-08:
A simple test would be to compute the center of mass based on non-background-color pixels. Do you want to try @GuillaumeFavelier ?

GuillaumeFavelier, 2021-03-08:
bounds are given by VTK's `ComputeVisiblePropBounds` right?

larsoner, 2021-03-08:
[Looks like it, yeah](https://github.com/mne-tools/mne-python/blob/main/mne/viz/backends/_pyvista.py#L989).

GuillaumeFavelier, 2021-03-08:
I'm on it

drammock, 2021-04-15:
Just found another corrupted 3D plot: https://mne.tools/dev/auto_tutorials/evoked/plot_20_visualize_evoked.html#d-field-maps

PR #9318 (2021-04-20): MNT: BackgroundPlotter has no attribute named image
This PR follows https://github.com/mne-tools/mne-python/pull/9313#issuecomment-822534247. For now, I'm trying to understand what is going on exactly.

Closes #9007
closed 2021-04-20 by larsoner