mne-tools/mne-python #9021: RFC: Re-applying baseline correction after ICA

hoechenberger, 2021-03-10:
Skimming over some Study Template analysis results with @agramfort, we found that some of our data appeared to be not baseline-corrected – some traces clearly didn't have a zero-mean during the baseline period.

So I started digging and figured that we had **indeed** applied baseline correction to our data; yet after cleaning the Epochs via ICA, the baseline signals of some channels had started floating away from zero.

Now this may not come as a surprise to some of you, but I have to admit it caught me off-guard. I'd never seen this behavior before with the data I'd worked with so far, and I've failed to reproduce a similarly visible effect using our `sample` data.

However, looking through the [ICA tutorial](https://mne.tools/dev/auto_tutorials/preprocessing/plot_40_artifact_correction_ica.html#fitting-and-plotting-the-ica-solution), where we clean Raw data, one can see the essence of this effect too (lower panel):

[image]

I'd like to propose and discuss the following changes and enhancements to deal with this behavior:

- Clearly state in the docs that baseline correction should be re-applied to Epochs and Evoked data after applying ICA
- `viz.plot_ica_overlay()` should automatically re-apply baseline correction to the cleaned signal if an Evoked was passed
    - for this, we'd need to add an `Evoked.baseline` attribute, which gets inherited from the parent `Epochs.baseline`. This has been on my agenda for a while anyway in order to improve provenance, so I'd might take a shot at this anyway.
- We could even go so far as to auto-reapply baseline correction to Epochs and Evoked data as part of `ICA.apply()`, so users wouldn't have to do this themselves. This behavior should be toggle-able using a new kwarg.

larsoner, 2021-03-10:
I think this is because of `pca_mean_`. Basically whatever mean value you have for each channel gets subtracted, then ICA unmixing/selection/mixing, then `pca_mean_` gets added back to the data. If you set `ica.pca_mean_[:] = 0`, does this problem go away?

I'm not actually 100% sure what the correct approach is here, but I don't like the fact that applying ICA imposes a DC shift in the data during an interval where the DC value of each channel is zero. This seems in theory incorrect, and in practice potentially dangerous.

larsoner, 2021-03-10:
I wonder if we should have a `center=True` (default) argument that says whether or not to use the empirical estimate of the mean for each channel to center the data. And I'm not 100% sure it makes sense to keep the default at `True`, we might want to (by deprecation) change it to `False`. If data that has been high-pass filtered (raw, or epochs based on high-passed raw data without baseline correction), then you probably don't want to do it since the high-passing should already center the data for you. If data that has been baseline corrected (epochs), it's probably not doing the right thing since the DC levels have been altered by the baselining, so your attempt to center with `pca_mean_` will be wrong (like this example).

Probably @agramfort has thought about this more than I have, though...

hoechenberger, 2021-03-10:
> If data that has been high-pass filtered (raw, or epochs based on high-passed raw data without baseline correction), then you probably don't want to do it since the high-passing should already [...]

If that's the case, we could simply decide to only center `if inst.baseline is not None`, I suppose? Would work right away with Epochs, and would require implementation of the `baseline` attribute for Evoked

hoechenberger, 2021-03-10:
> If that's the case, we could simply decide to only center `if inst.baseline is not None`, I suppose? 

... plus a check whether the data has been high-passed, of course …

larsoner, 2021-03-10:
That seems reasonable to me, and we could still pair it with a `center='auto' | True | False` where `'auto'` centers if it makes sense to do so (data not high-passed, not baseline corrected), where `center` defaults to `True` in 0.23 and goes to `'auto'` in 0.24 via deprecation. But let's wait to hear if @agramfort sees this as conceptually correct.

In the meantime you can work on the evoked baseline attribute again if you want :)

larsoner, 2021-03-10:
SSP is a linear spatial operator, so if all channels have zero mean (during some interval) all channels will have zero mean after the operation (in that same interval). This is because the mean (over that interval) is also a linear operator.

To demonstrate how ICA violates this, let's say we have two channels of data for one 3-sample epoch that that each already have zero mean over time, for example `[[1, 0, -1], [0, 0, 0]]`. If the empirical means estimated by ICA during fitting happened to be, say -1 and 0, ICA will:

1. Subtract the empirical estimated means, giving `[[2, 1, 0], [0, 0, 0]]`
2. Apply the ICA operator, which is just some linear spatial operator. Let's pick one that just swaps data between channels, `[[0, 1], [1, 0]]` (ICA in practice will do something different like unmix-exclude-mix, but this gives a simple example to play with). So now our data after the ICA operator is `[[0, 0, 0], [2, 1, 0]]`.
3. Add the means back, giving `[[-1, -1, -1], [2, 1, 0]]`. 

Now our previously-zero-mean channels have means `-1` and `1`, which is not good. Let's call this operation `B = ICA(A)`. If we apply this operation to identical data, that will (trivially) have `B = ICA(A)` having means `-1` and `1`, too.

And this is where you can see that the `ICA` operation above is not linear. We know that `B + B = ICA(A) + ICA(A)` but that this means `2B = ICA(A) + ICA(A)`. We know that since `B` has means `-1` and `1`, `2B` will have means over time of `-2` and `2`. For linearity to hold, we need that `2B = ICA(A) + ICA(A) = ICA(A + A) = ICA(2A)`, but unfortunately `ICA(2A)` will go from `[[2, 0, -2], [0, 0, 0]]` to start, through step 1 to `[[3, 0, -1], [0, 0, 0]]`, through step 2 to `[[0, 0, 0], [3, 0, -1]]`, through step 3 to `[[-1, -1, -1], [3, 0, -1]]`, which has means `-1` and `2/3`, which do not equal `-2` and `2`. Hence our ICA operation is not a linear operator, which I think it is meant to be...

If we remove the mean subtraction at the beginning and mean re-addition at the end, ICA will be a linear spatial operator (just like SSP) in the end. To me this is starting to seem like the correct option.

cbrnr, 2021-03-10:
I fail to see why ICA should estimate means that are not equal to the sample means, in which case your example doesn't work.

larsoner, 2021-03-10:
> I fail to see why ICA should estimate means that are not equal to the sample means, in which case your example doesn't work.

Indeed the example is contrived but there are cases where it can happen. Simple examples like estimating / fitting on some or all segments of raw and applying to epochs, or estimating/applying on different segments of raw data, or estimating/applying on epochs/raw, etc. will all have this problem to varying degrees, depending on the data characteristics. The severity of the problem is going to depend on the difference in the means of the data it's fit on and the data it's being applied to. Having baseline corrected epochs is a particularly problematic case (@hoechenberger's above). The key is that the estimated sample mean from PCA needs to accurately reflect the true underlying mean of the data you're applying the ICA to.

So IMO the particularly problematic case is estimation on raw, then application on baseline corrected epochs. I think if you fit and apply to the same epochs (baseline corrected or not), or fit and apply to raw, or fit to raw and apply to non-baseline-corrected epochs (assuming you have enough raw samples, etc.) and then optionally baseline correct, things are probably okay.

cbrnr, 2021-03-10:
I agree, but this isn't really an ICA problem, but it happens because we've added an explicit zero-meaning step. I'm not even sure why we're doing this, because PCA could be computed on the covariance matrix, but we're probably using SVD. So I guess what I'm trying to say is that we probably shouldn't make this the default and leave centering to the user (but of course highlight this in the docs).

agramfort, 2021-03-10:
btw I just ask Arnaud Delorme and EEGLAB also does the centering.

Now if it turns out to help we could add an `assume_centered` book param in
_ICA class
to possibly skip centering.

More soon

>

hoechenberger, 2021-03-11:
I've created the following MWE to check for interactions between

- baselining and not baselining the epochs we remove ICs from
- baselining and not baselining epochs before ICA **decomposition**
- centering and not centering the signal via `ICA.pca_mean_`

```python
import os
import itertools
import mne
from mne.preprocessing import ICA

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
event_id = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
            'visual/right': 4, 'face': 5, 'button': 32}

raw = (mne.io.read_raw_fif(sample_data_raw_file, preload=True)
       .filter(l_freq=1., h_freq=None))
events = mne.find_events(raw, stim_channel='STI 014')

for baseline_data, baseline_ica, center_pca in itertools.product([True, False],
                                                                 [True, False],
                                                                 [True, False]):

    epochs_data = mne.Epochs(raw, events, tmin=-0.2, tmax=0.5,
                             event_id=event_id, preload=True, baseline=None)
    epochs_ica = epochs_data.copy()

    if baseline_data:
        epochs_data.apply_baseline()
    if baseline_ica:
        epochs_ica.apply_baseline()

    ica = ICA(n_components=15, random_state=97)
    ica.fit(epochs_ica)
    if not center_pca:
        ica.pca_mean_[:] = 0

    evoked = epochs_data.average()
    fig = ica.plot_overlay(evoked, exclude=[0])
    title = (f'data_'
             f'{"baselined" if baseline_data else "not_baselined"}-'
             f'ica_'
             f'{"baselined" if baseline_ica else "not_baselined"}-'
             f'{"centered" if center_pca else "not_centered"}')
    fig.suptitle(title)
    fig.tight_layout()
    fig.savefig(f'{title}.png')
```

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

hoechenberger, 2021-03-11:
Since the effect is most pronounced for EEG data, here are the EEG plots again:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

hoechenberger, 2021-03-11:
What we're now doing in the Study Template is we don't apply baseline correction to anything until **after** the data has been cleaned. The results we get that way look sane, as far as I can tell.

cbrnr, 2021-03-11:
ICA cannot handle DC well, so it is recommended to high-pass the data (cutoff e.g. 0.5Hz or 1Hz). This means that baselining could even have detrimental effects, i.e. if all concatenated epochs already have zero mean, baseline-correcting might introduce an offset.

agramfort, 2021-03-11:
we converge to the same thing.

Do not baseline epochs passed to ICA.

Apply baseline only after doing ICA cleaning.

No need to enable not centering the PCA step.

>

hoechenberger, 2021-03-11:
I will update the docs accordingly.

PR #9033 (2021-03-30): Recommend to baseline after (but not before) ICA; and warn otherwise; warn if data was not high-passed when fitting ICA
Closes #9021
commit 014e9526 (2021-03-30): Recommend to baseline after (but not before) ICA; and warn otherwise; warn if data was not high-passed when fitting ICA (#9033)

* Recommend to baseline after, but not before, ICA, and warn otherwise

Closes #9021

* Add changelog entry

* Adjust tests

* Rephrase warnings & notes, add new warning if data not hi-passed

* Fix tests to handle high-pass filter warning

* Fix another test and ICA tutorial

* Rephrase warning as suggested by @agramfort

* Add tests for new warnings

* Add docstring to test

* Do not baseline correct Epochs in tutorial [circle full]

* Move versionchanged and note  [... cut]
closed 2021-03-30 by hoechenberger