mne-tools/mne-python #9022: viz.plot_compare_evokeds() calculates RMS for all channel types, but claims to be calculating GFP

hoechenberger, 2021-03-10:
This is a similar issue to the one fixed via #8775

`viz.plot_compare_evokeds()` should calculate GFP for EEG and RMS for MEG data. The `combine='gfp'` option should remain unchanged and simply pick the correct calculation depending on the channel type(s). This means that in case of M/EEG data, we'd have to apply different `combine` functions per subplot, so this is not a totally non-trivial change as far as I can tell (but might be wrong).

matteoanelli, 2021-03-17:
I can work on this!

hoechenberger, 2021-03-17:
This is actually not a super trivial issue and might require a bit of refactoring the current plotting code. I can give you a brief summary hopefully later today

PR #12410 (2024-02-02): actually use GFP for EEG channels in plot_compare_evokeds
fixes #9022
closed 2024-02-02 by drammock