mne-tools/mne-python #9025: Optimize head placement in viz.plot_evoked() when spatial_colors=True

hoechenberger, 2021-03-11:
This is what I'm seeing in one of our analyses:
[image]

Would be great if the head could be automatically placed such that it won't overlap with the traces – maybe even in a separate subplot?

drammock, 2021-03-11:
I think we rely on matplotlib's algorithm for picking the "best" position, but I think it might only consider corners, not sides.  how does it look without the GFP, does it still fail then, or does it use one of the bottom corners?

I'd like to avoid implementing our own "find the best spot for this" algorithm.  I'm not wild about the separate subplot idea (because in most cases it will lead to a lot of unnecessary whitespace); I'd rather

1. make it easier for a user to tell us where they want the head circle, or
2. make it really easy to add it after the fact.

larsoner, 2023-10-01:
I'd like to add a third option which is to let people tweak the plot afterward if they need to. It requires learning some stuff about matplotlib internals (axes, parasite axes, etc.) but I think if we add it to some tutorial or example it'll be discoverable enough!

jona-sassenhagen, 2025-06-02:
@drammock 

>I think we rely on matplotlib's algorithm for picking the "best" position, but I think it might only consider corners, not sides. how does it look without the GFP, does it still fail then, or does it use one of the bottom corners?

This is the logic:

```def _handle_spatial_colors(...
...
    loc = 1 if psd else 2  # Legend in top right for psd plot.
    _plot_legend(pos, colors, ax, bads, outlines, loc)
```

jona-sassenhagen, 2025-06-02:
The dirty fix is to use `ts_args` for `plot_joint` to change the ylim