mne-tools/mne-python #9030: Adding new connectivity indexes

cmista, 2021-03-12:
#### Describe the new feature or enhancement
The simplest way to check the relation between two signals is using correlation or a similar a similar approach. However, there are connectivity indexes that don't need to make the assumption of linearity between the interaction of the signals. These kind of indexes can be developed based on the information theory.  

#### Describe your proposed implementation
New functions in the Connectivity Estimation group would be nice to be incorporated. 

#### Describe possible alternatives
Mutual information is a good option as a non-directed measure.
Transfer entropy seems to be a good option for a directed measure (causal) - although this most likely is too complex to handle myself alone  

#### Additional comments
A review about connectivity indexes
Bastos André M., Schoffelen Jan-Mathijs. A Tutorial Review of Functional Connectivity Analysis Methods and Their Interpretational Pitfalls. Frontiers in Systems Neuroscience  

Transfer entropy paper
Thomas Schreiber. Measuring Information Transfer. https://arxiv.org/abs/nlin/0001042v1

A nice article about information theory in neuroscience
Nicholas M. Timme and Christopher Lapish. A Tutorial for Information Theory in Neuroscience. Eneuro

agramfort, 2021-03-12:
is there existing python code for this already?

>

cmista, 2021-03-12:
Yes, there are for python and for Matlab. Probably the word "new" in the tittle is confusing.

agramfort, 2021-03-12:
can you provide links to it?

drammock, 2021-03-12:
@cmista next steps are:

1. if there is existing python / matlab code, give links to it and make sure its license is one that allows us to adapt / incorporate it (MNE-Python is BSD license)
2. propose what the API for this would look like (what would be the function/method name, what would its arguments be, what would the return value look like)

cmista, 2021-03-13:
> @cmista next steps are:
>
>     1. if there is existing python / matlab code, give links to it and make sure its license is one that allows us to adapt / incorporate it (MNE-Python is BSD license) [...]

Thank you for clarifying the steps. 

Point 1
I propose to start with the Mi, and surely trying the sklearn implementation (https://scikit-learn.org/stable/modules/generated/sklearn.feature_selection.mutual_info_regression.html) using channels for features.

I been looking for TE through the toolboxes, pondering license, support, and features. Best candidate is Python IDTxl (https://github.com/pwollstadt/IDTxl). The toolbox is open, but couldn't find the terms of use agreement. The toolboxes based on Matlab (MuTE and TRENTOOL https://github.com/trentool/TRENTOOL3) are GPL, but they seems to require a LOT of effort to be adapted.

Point 2  
Working on that.

drammock, 2021-03-13:
For mutual info, wrapping scikit-learn should be fine, it is already a dependency for some of our functionality. 

ITDxl has [a license file](https://github.com/pwollstadt/IDTxl/blob/master/LICENSE) in the root of their repository.  It is GPL licensed, which means we can't adapt or incorporate it without getting permission from the author(s) first.

jshanna100, 2021-03-15:
I made an implementation of directed Phase Transfer Entropy (dPTE) last year, and have been wanting to integrate it with MNE Python for a while, but could never find the time. I would be happy to contribute my code as the starting point - it would also be a good opportunity to make sure everything works correctly.

agramfort, 2021-03-15:
can you point to your implementation? just to see how long and complex
the code is? just curious :)

jshanna100, 2021-03-15:
Yes, gladly. It's currently buried here in an otherwise very disorganised repo:
[https://github.com/TinnErlangen/ATT/blob/master/cnx/dPTE.py](url)

The main function, epo_dPTE, takes an epoch array of data [n,c,t], where n is number of epochs, c is channels, and t is time, and produces [n,c,c] directed connectivity matrices for the desired frequency range. It already has parallel support, and also allows one to choose wavelet or hilbert transforms for instantaneous phase estimation.

The implementation is based on Lobier et al, 2014, Neuroimage, and Hillebrand et al 2016, PNAS.

agramfort, 2021-03-16:
ok it's based on multivariate binning.

agramfort, 2021-03-17:
Honestly I would say we need some evidence that this should be used by MNE users. In a word before adding this I feel we need "the why". I would like to avoid the argument that it should be added to MNE just but it’s not yet in MNE. Is there some example, some paper, some dataset that would be convincing?

my 2c

cmista, 2021-03-18:
Good point @agramfort , because it give rise to 2 questions. One is whether it's worth studying nonlinear coupling in brain activity (which I'm biased to say yes). The second point is whether this feature will be useful for MNE-python users. I guess some users might be confused with so many indexes, and they could ending p-hacking their results in the worst-case scenario. I'll leave it as it, and pick another issue.

jshanna100, 2021-03-18:
The primary benefit in my view is the addition of _directed_ connectivity to the MNE repertoire. As far as I know, there are still only non-directed methods available in MNE. Secondarily, it would be nice to offer an information theory based method.

As far as papers go, I think a case is already more than adequately made in the two citations I gave above. I would personally name dPTE's robustness against signal leakage as its primary, distinctive strength.

Stepping back a bit, I have thought for a while that explicitly comparing available connectivity measures on the same dataset would make an excellent example/tutorial - preferably using something like synthetically generated coupled oscillators (e.g. kuramoto or neural mass models) as a sort of ground truth.

agramfort, 2021-03-19:
I like what you say Jeff ! let's give this a try then. I am super curious
to see who the benchmark on synthetic data will reveal !

>