mne-tools/mne-python #9031: sidebar directive in the new doc theme not producing a sidebar

hoechenberger, 2021-03-12:
It appears `.. sidebar::` will not actually create a sidebar anymore – which **does** make sense IMHO with the new layout; however maybe we should replace "sidebars" with "info boxes" now?

Source:
[image]

Rendering:
[image]

larsoner, 2021-03-12:
Can you look to see what needs to be changed in https://github.com/larsoner/sphinx-bootstrap-divs which we vendor in https://github.com/mne-tools/mne-python/tree/main/doc/sphinxext/sphinx_bootstrap_divs ?

larsoner, 2021-03-12:
... actually we shouldn't even really use that anymore since it's for sphinx-bootstrap-theme, which uses BS3, and pydata-sphinx-theme uses BS4 I think.

I'd propose adding whatever the correct thing is (if it is info boxes) to the upstream pydata-sphinx-theme to see if they'd want it. If they do then let's implement it there, if they don't then we can vendor in our own `doc/sphinxext/*` somewhere.

drammock, 2021-03-13:
> actually we shouldn't even really use that anymore since it's for sphinx-bootstrap-theme, which uses BS3, and pydata-sphinx-theme uses BS4 I think.

FYI I updated `doc/sphinxext/sphinx_bootstrap_divs` when we switched themes, to be BS4 compliant.  (not certain I did it fully or in the best way, but I at least made it work).

The real problem here was that the sidebars used to be put in a `div` and now they're in an `aside` tag instead, and our CSS selector was too specific.  I've pushed a fix as part of #9036.

hoechenberger, 2021-03-13:
Thanks @drammock!

commit d309e3dc (2021-03-14): MRG, DOC: Bump theme version (#9036)

* bump theme version

* use theme support for custom quicklinks (discourse)

* add discord icon/link

* update CSS variable names with new theme prefix

* use system font stacks as fallback

* remove webfonts

* update font versions for circle build

* fix sidebar styling (fixes #9031)

* fix: handle sidebars that are divs or asides
closed 2021-03-14 by agramfort